react-native-camera/react-native-camera #2601: react-native-camera getting permission error in windows uwp

SelvaraniAlagumuthu, 2019-11-21:
Hi,

I try to use react-native-camera for qrcode scanner. i did react-native link. Its working fine in android. while i try to windows uwp. i am getting react-native-permissions error in uwp. Could you elabrate your readme for windows also.

cristianoccazinsp, 2019-11-27:
Pardon my ignorance, but what are you using exactly to run react native apps on UWP? Just curious.

stmoy, 2020-01-11:
I expect that this issue will be solved with #2669?

stmoy, 2020-03-11:
As of #2723, we've updated this module to work with the latest version of RN for Windows. Please try it out and let me know if you're still hitting an issue.

mudaligekosala, 2020-08-23:
Is react-native-camera doesn't work on windows simulator without webcam ?

stmoy, 2020-08-24:
CC: @chrisglein 

Let me follow up internally to see what's up.

stmoy, 2020-08-24:
@mudaligekosala - can you confirm your setup? Is this an app running in a Windows VM/simulator, or on a physical Windows machine? If the former, does the Camera app work in the VM? If the latter, does the device have a webcam?

mudaligekosala, 2020-08-25:
@stmoy - I am running using windows machine.but PC doesn't has a webcam.I build the Standalone application and installed it in the machine with webcam still showing the pending screen.
Thanks for following up

jonthysell, 2020-08-25:
What version of RN (and RNW) are you using, and what exactly are you trying to build? Your own app? A sample?

We have a sample app here: https://github.com/microsoft/react-native-windows-samples/tree/master/samples/CameraDemo

Can you tell me if that builds/works for you?

stmoy, 2021-01-06:
CC @mudaligekosala - have you had a chance to see Jon's notes above?

closed 2021-03-14 by stale[bot]