react-native-camera/react-native-camera #2607: Android low quality after takePhoto

teriologia, 2019-11-25:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?
- [x] Try different devices
- [x] Try to set quality option to 1.0

**Platforms**

ANDROID

**Versions**

- Android: 6-9
- iOS:?
- react-native-camera: 3.11.1
- react-native: 0.59.9
- react: 16.8.6

**Description/Current Behaviour**

I've problem with quality I just update RNCamera package to 3.11.1 from 3.8.0 everything works but when I take a photo I'm getting low quality result. in my app peoples should read article from taken photo but after this update this isn't possible with this quality

Before take a photo(preview):
[image]

After take a photo (result):
[image]

my camera options ( I also try quality: 1):

> quality: 0.7,
  width: 720,
  orientation: RNCamera.Constants.Orientation.portrait,
  base64: true,
  pauseAfterCapture: true,
  // skipProcessing: false,
  doNotSave: false, 
  // /** iOS only */
  forceUpOrientation: true,
  writeExif: true,

cristianoccazinsp, 2019-11-25:
Have you tried with higher quality? Quality 0.7 and width 720 is very low, either use 720 with quality 1 or  a higher width and lower quality. 720 width is very small I think.

teriologia, 2019-11-26:
> Have you tried with higher quality? Quality 0.7 and width 720 is very low, either use 720 with quality 1 or a higher width and lower quality. 720 width is very small I think.
> 
> Update: I think that the difference you are seeing is because the quality parameter was previously not being passed to the camera at all, so the camera would always capture uncompressed JPEGs. Now the quality parameter will in fact be passed to the camera for the JPEG compression.

I already try to set quality 1.0 but same result. This bug appears when I update camera package from 3.8.0 to 3.11.0 whats different everything was perfect at 3.8.0

cristianoccazinsp, 2019-11-26:
There should be no difference. Have you tried removing the quality param
completely?

What happens if you use a higher width? You probably want to match the
screen's width in pixels (usd the pixel conversion helpers from RN).

teriologia, 2019-11-27:
> There should be no difference. Have you tried removing the quality param completely? What happens if you use a higher width? You probably want to match the screen's width in pixels (usd the pixel [...]

When I set

quality: 0.5,
width: 1920

everything perfect but this time my base64 code's size is very high 

quality: 0.7
width: 720
Base64 Size: 90-110kb
I was using older version since 2 months with and everything was good

now I'm usig this settings and my base64 size is higher
quality: 0.5
width: 1920
Base64 Size: 500- 550kb

cristianoccazinsp, 2019-11-27:
What device are you using? I've tried your setup with 720 and quality 1 and
I see a pretty good picture, even with quality 0.7 or 0.9.

I wonder if the quality loss you are seeing is because of the native JPEG
compression passed to the camera API call. Maybe your device was previously
capturing raw photos and only converted to JPEG when resizing.
If this is true, you were previously loading s very large image into
memory. Did you notice any difference in capture speed?

If you are willing to test a PR I can try making one so that you can
disable the quality prop being passed to the camera capture call and only
used when resizing.

sibelius, 2019-11-27:
is this related to `skipProcessing` flag?

cristianoccazinsp, 2019-11-27:
Not really since the code path is the same when requesting width and
base64. With skip processing those two would have been ignored.

teriologia, 2019-11-27:
> What device are you using? I've tried your setup with 720 and quality 1 and I see a pretty good picture, even with quality 0.7 or 0.9. I wonder if the quality loss you are seeing is because of the [...]

My device: LG V30plus
When I delete quality prop everything is perfect but I just notice `autoFocus='on'` didn't work when I delete autoFocus my image looks good and base64 code size is normal
solition: delete quality, increase width, remove autoFocus because this option didn't works

I try almost 4 different phone 
lg V30plus
samsung j7
iphone 6
LG very old one I don't remember model

cristianoccazinsp, 2019-11-27:
So do you see the same quality issue on iphone 6? 

I've been trying to test it, but pictures are crystal clear even with `width=720` and `quality=1`

If this is a device specific issue, you could try editing this line [here](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L680) on the java code and see if there are any changes. Perhaps the native camera JPEG compression is doing something odd to your pictures. If that's true, we probably want a prop to disable it and only use the slow Java bitmap compression.

If you only want the picture to be big enough to fit the screen, you should not hardcode a 1920 or 720 width. You could use something like this to get the exact pixel size you need for your screen:

```
import {Dimensions, PixelRatio} from 'react-native';
let {width, height} = Dimensions.get('window');
let pictureWidth = PixelRatio.getPixelSizeForLayoutSize(width);
```

teriologia, 2019-11-27:
> So do you see the same quality issue on iphone 6?
>
> I've been trying to test it, but pictures are crystal clear even with `width=720` and `quality=1`
>
> If this is a device specific issue, you could [...]

Thanks for help actually this issue causing by width: when I change my code 
```
import {Dimensions, PixelRatio} from 'react-native';
const {width, height} = Dimensions.get('window') * 1.5;
```
everything look good 150-170kb base64 size and normal quality photos

cristianoccazinsp, 2019-11-28:
So did you fix the issue by just using `screen width * 1.5`?  Sounds a bit odd.

teriologia, 2019-11-29:
> So did you fix the issue by just using `screen width * 1.5`? Sounds a bit odd.

I don't understand but everything works good now. My photos looks normal

closed 2020-02-04 by stale[bot]