react-native-camera/react-native-camera #2608: Image not displayed after a

ahmadallo1, 2019-11-25:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-11-25:
android or ios?

ahmadallo1, 2019-11-25:
@sibelius the issue is on IOS

cristianoccazinsp, 2019-11-25:
What do you mean exactly by upgrading the app?

If you are using iOS, you should NOT be storing any URI returned by any library. IOS will change part of the path after each application update or backup restore. Not really related to the camera library.

ahmadallo1, 2019-11-26:
@cristianoccazinsp I mean updating the application. Is there a way to get those images?

ahmadallo1, 2019-11-26:
@cristianoccazinsp I mean updating the application. Is there a way to get those images again?

cristianoccazinsp, 2019-11-26:
The images will remain there, but you need to adjust your stored path to
the new path apple will assign to your app after an update.

I recommend you check the react native fetch blob library to work with the
file system.

cristianoccazinsp, 2019-11-26:
If your goal is to permanently store the image on the device, you will most likely need to do two things. First, copy the image to the app's documents storage or somewhere that's not the cache folder (which is where RN stores the pictures, and this folder can be cleaned up automatically). Second, only store the relative path, and re-build it at run time when needed. That is, store only the part after `fs.dirs.DocumentDir`, and prepend it when loading it from your storage.

ahmadallo1, 2019-11-26:
@cristianoccazinsp I want to store the images on the device in order to display them again. So first we are not able to get saved images from the cache folder?

cristianoccazinsp, 2019-11-26:
You are, you can also use `fs.dirs.CacheDir` (from react-native-fetch-blob) to find where's the cache folder. You could do the same I described above; only store the relative path, and then prepend `fs.dirs.CacheDir` to re-build the final path. If your app is updated and Apple changes your app's location, you would still get the right path/data.

However, the cache folder can be cleaned up by the OS at any time (both android and iOS), so it is not a safe place to store permanent things.

sibelius, 2019-11-26:
I think this should be in the docs

cristianoccazinsp, 2019-11-26:
This is definitely not a RNCamera issue though, just file system management of each platform. For iOS, it is quite surprising apple changes the storage path so frequently.

closed 2020-01-27 by ahmadallo1