react-native-camera/react-native-camera #2609: After calling takePictureAsync, I get error Image could not be captured【iOS】

kettyy, 2019-11-27:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:12.4.1
- react-native-camera:3.11.1
- react-native:0.61.5
- react:16.9.0
  

**Description/Current Behaviour**
On iOS, after calling takePictureAsync, I get error Image could not be captured

**Expected Behaviour**
Capture image after takePictureAsync is called

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-11-27:
Can you provide any code sample? What device? Can you get the actual error from XCode?

kettyy, 2019-11-28:
> Can you provide any code sample? What device? Can you get the actual error from XCode?

1. Device is iPhone6.
2. I can't get the actual error from XCode.
3. The sample code is as follows:
```
export default inject('globalStore')(
  observer(({navigation, globalStore}: Props) => {
    const theme = useTheme();

    let camera: any = useRef(null);

    const {getAppLocale} = globalStore;

    const [canDetectFaces, setDetect] = useState(false);
    const [isFace, setIsFace] = useState(false);

    const takePicture = async ({faces}: any) => {
      if (Array.isArray(faces) && faces.length && camera) {
        if (faces.length === 1) {
          const [face] = faces;
          if (face.bounds.size.width / Dimensions.get('window').width > 0.4) {
            setIsFace(true);

            const options = {
              forceUpOrientation: true,
              fixOrientation: true,
              quality: 0.8,
            };
            
            camera
              .takePictureAsync(options)
              .then((err: any) => {
                console.log(err);
              })
              .catch((err: any) => {
                console.log(err);
              });

            //   faceStore.handleSubmit(data, navigation.goBack)
          } else {
            Methods.showToast(getAppLocale['settings.face.no.face'], {
              position: Toast.positions.BOTTOM,
              duration: 1000,
            });
          }
        } else {
          Methods.showToast(getAppLocale['settings.face.only.one'], {
            position: Toast.positions.BOTTOM,
            duration: 1000,
          });
        }
      }
    };

    return (
      <SafeAreaView
        style={{
          flex: 1,
          backgroundColor: theme === 'dark' ? '#1A1B1A' : '#fff',
          position: 'relative',
        }}>
        <RNCamera
          ref={ref => {
            camera = ref;
          }}
          style={styles.camera}
          {...(!isFace && {
            faceDetectionMode: RNCamera.Constants.FaceDetection.Mode.accurate,
            onFacesDetected: canDetectFaces ? takePicture : undefined,
          })}
          captureAudio={false}
          onMountError={() => {
            Methods.showToast(getAppLocale['settings.face.permissions']);
            navigation.goBack();
          }}
          onCameraReady={() => {
            setDetect(true);
          }}
        />
      </SafeAreaView>
    );
  }),
);
```

cristianoccazinsp, 2019-11-28:
Have you tried without face detection?

kettyy, 2019-11-29:
> Have you tried without face detection?
> 
> I think you might be getting an error for adding/removing face detection capabilities and capturing at the same time.

Thank you very much for your help, I'm sure the reason for `setIsFace (true)` is asynchronous, I will close this issue

closed 2019-11-29 by kettyy