react-native-camera/react-native-camera #2610: Is it possible to check if Camera Android Permission is visible?

ianvasco, 2019-11-27:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Hi guys. I'm wondering if it is possible to check if the android permission dialogue is visible when requested from the library. I need to make a state change when any of these dialogues are visible. Is it possible?

**Tags**
question

cristianoccazinsp, 2019-11-27:
The only work around I can think of is to use the `pendingAuthorizationView` component and do something there. You know if the component has been mounted for some time it is because it is waiting for authorization.

Otherwise, check for permissions manually before mounting the camera?

ianvasco, 2019-11-27:
I'm checking for permissions manually but still I need to manage when user deny it and then tries to open de camera. Could you please tell me where is this component `pendingAuthorizationView`?

cristianoccazinsp, 2019-11-27:
It's on the docs, but here's an example:

https://github.com/react-native-community/react-native-camera/blob/master/examples/advanced/advanced/src/screens/Camera.js#L617

https://github.com/react-native-community/react-native-camera/blob/master/examples/advanced/advanced/src/screens/Camera.js#L622

ianvasco, 2019-11-27:
It worked like a charm, thanks!

closed 2019-11-27 by ianvasco