react-native-camera/react-native-camera #2612: Hardware issue from customer

teriologia, 2019-11-29:
# Question

**To Do First**
- [x ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Hi I'm using RNCamera in my product but I'm getting hardware errors from my customers in Sentry like this

> LegacyExceptionUtils.java line 73
android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError

> Camera2.java line 970
com.google.android.cameraview.Camera2.chooseOptimalSize

My question is: Can RNCamera is cause this issue?
not every customer has this issue only this devices crashes:
SM-G930F
SM-J730F
Galaxy Tab A 10.1

sibelius, 2019-11-29:
can you provide the full stacktrace?

cristianoccazinsp, 2019-11-29:
Looks like a Camera2 issue. If you're able to reproduce the issue, can you test with Camera1?

Samsung devices are well known to cause issues with pretty much everything though, and my short testing with Camera2 showed that not even modern but cheap devices support it.

teriologia, 2019-11-29:
> Update: this code might be related

I'll try with camera1 next release but is there any performance loss in camera?

cristianoccazinsp, 2019-11-29:
No, Camera1 in fact works faster and is more stable. The only drawback is that it doesn't have access to advanced features (that are not used by RNCamera anyways), and is considered "deprecated" so who knows when google will decide to shut it down completely. 

Still, can you provide the full stack trace so we know what's going on?

teriologia, 2019-11-29:
> can you provide the full stacktrace?

sure.
[Camera2Api.txt](https://github.com/react-native-community/react-native-camera/files/3905630/Camera2Api.txt)

teriologia, 2019-11-29:
> No, Camera1 in fact works faster and is more stable. The only drawback is that it doesn't have access to advanced features (that are not used by RNCamera anyways), and is considered "deprecated" so [...]

Thanks for info I'll try here my full stacktrace:
[Camera2Api.txt](https://github.com/react-native-community/react-native-camera/files/3905636/Camera2Api.txt)

teriologia, 2019-11-29:
> No, Camera1 in fact works faster and is more stable. The only drawback is that it doesn't have access to advanced features (that are not used by RNCamera anyways), and is considered "deprecated" so [...]

one last question can I make barcode reader with camera1 API

cristianoccazinsp, 2019-11-29:
Yes, of course!

cristianoccazinsp, 2019-11-29:
Leaving the relevant parts of the stack trace here.... Not really sure what's going on, but the crash looks like started here: 
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L97
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L895

If I had to guess, `Fatal Exception: java.lang.IllegalArgumentException: Surface was abandoned` tells me it was caused by race conditions just like the ones that have been getting fixed on Camera1. 

The code most likely needs synchronization across start/stop calls, and possibly on some configuration changes; just like Camera1 needed.

@sibelius  is there anywhere on the docs that recommends the use of Camera2? It definitely needs some more love to make it as good as Camera1. Seeing how some devices are not even implementing Camera2 yet I don't think it will go away any time soon.

Device: SM-J730F (Android 9)

```
#0. Crashed: main
       at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize + 84(SurfaceUtils.java:84)
       at android.hardware.camera2.params.OutputConfiguration.<init> + 281(OutputConfiguration.java:281)
       at android.hardware.camera2.params.OutputConfiguration.<init> + 145(OutputConfiguration.java:145)
       at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession + 513(CameraDeviceImpl.java:513)
       at com.google.android.cameraview.Camera2.startCaptureSession + 895(Camera2.java:895)
       at com.google.android.cameraview.Camera2$1.onOpened + 97(Camera2.java:97)
       at android.hardware.camera2.impl.CameraDeviceImpl$1.run + 148(CameraDeviceImpl.java:148)
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 214(Looper.java:214)
       at android.app.ActivityThread.main + 7156(ActivityThread.java:7156)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run + 494(RuntimeInit.java:494)
       at com.android.internal.os.ZygoteInit.main + 975(ZygoteInit.java:975)

Fatal Exception: java.lang.IllegalArgumentException: Surface was abandoned
       at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize + 84(SurfaceUtils.java:84)
       at android.hardware.camera2.params.OutputConfiguration.<init> + 281(OutputConfiguration.java:281)
       at android.hardware.camera2.params.OutputConfiguration.<init> + 145(OutputConfiguration.java:145)
       at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession + 513(CameraDeviceImpl.java:513)
       at com.google.android.cameraview.Camera2.startCaptureSession + 895(Camera2.java:895)
       at com.google.android.cameraview.Camera2$1.onOpened + 97(Camera2.java:97)
       at android.hardware.camera2.impl.CameraDeviceImpl$1.run + 148(CameraDeviceImpl.java:148)
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 214(Looper.java:214)
       at android.app.ActivityThread.main + 7156(ActivityThread.java:7156)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run + 494(RuntimeInit.java:494)
       at com.android.internal.os.ZygoteInit.main + 975(ZygoteInit.java:975)
```

closed 2020-02-04 by stale[bot]