react-native-camera/react-native-camera #2615: [ASK] Video Camera on Low Quality

michael-harley, 2019-12-01:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
How can I post-process my video? I already get the lowest possible quality on Android, that is 480p or 720x480.
But I want to get 480x360 or lower if possible, as I don't really need good quality.

I need to get the lowest possible size of the video, as my project only need videos for evidence and doesn't really need a good quality (takes too much storage space).

Any possible workaround for this?
All answer will do help! Thanks in advance.

**Tags**
#videoQuality 

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-12-01:
For iOS, there's also 288p (CIF) which is very, very low quality. I'm not sure why this hasn't been implemented for Android, but it shouldn't be too hard to do so (constant is already there https://developer.android.com/reference/android/media/CamcorderProfile.html#QUALITY_CIF).

This quality lets you get videos of about 2 to 3 minutes in 20mb size.

michael-harley, 2019-12-02:
> For iOS, there's also 288p (CIF) which is very, very low quality. I'm not sure why this hasn't been implemented for Android, but it shouldn't be too hard to do so (constant is already there https://developer.android.com/reference/android/media/CamcorderProfile.html#QUALITY_CIF).
> 
> This quality lets you get videos of about 2 to 3 minutes in 20mb size.
> 
> There are probably other ways to achieve even lower quality/size, by reducing frame rate for example, but I'm not sure any of those are exposed to RNCamera, yet.

Yes, iOS has the lowest quality of 288p

But my project is using Android as Android phones are cheap.
And there might be 100-200 videos each day.

1 video only contain of 2 secs, that is 1MB for each video with the lowest quality.

And 100-200MB each day. With 20GB or so space. If possible I want to compress the quality even more.

If you have a workaround for this, please do share it with me 🙏.
Thanks!

Outsource libraries are welcomed. :)

closed 2020-02-07 by stale[bot]