react-native-camera/react-native-camera #2620: Component RNCamera re-registered bubbling event 'topPictureTaken' as a direct event

RixBai, 2019-12-03:
What's the solution?

cristianoccazinsp, 2019-12-04:
Can you provide more info?

RixBai, 2019-12-04:
I installed both react-native-document-scanner and react-native-camera package and I got this error.
When I deleted the react-native-document-scanner package it worked.
Can you please let me know how to resolve this problem?

cristianoccazinsp, 2019-12-04:
Can't really help with more details. What platform this this happen on? What are you doing exactly?

However, it seems like the error is related to both libraries using the same event name for `onPictureTaken`. Probably both libraries' fault to not properly namespace their events under their module name :(. If that's the case, you probably can't do much about it, except ask for a change on that event name on any of the libraries.

RixBai, 2019-12-04:
I see.
FYI, I am getting the error on iOS.

cristianoccazinsp, 2019-12-04:
Conflict right here:

https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L30

https://github.com/Michaelvilleneuve/react-native-document-scanner/blob/master/ios/DocumentScannerView.h#L6

Probably a similar issue on Android.

A similar issue was reported here: https://github.com/react-native-community/react-native-video/pull/1625

Their fix was to have the library replace the event type. You might have luck replacing the event on https://github.com/Michaelvilleneuve/react-native-document-scanner/blob/master/ios/DocumentScannerView.h#L6 , but no idea if it would still work. If anything, react-native-document-scanner would probably benefit by replacing the event with the direct event type instead.

RixBai, 2019-12-04:
I see.
Let me try to replace the event.
Thank you, @cristianoccazinsp

fuchien, 2020-01-26:
I got the same problem in iOS.
"react-native": "0.61.2",
"react-native-camera": "^3.15.1",
"@woonivers/react-native-document-scanner": "^2.1.1".
@RixBai, did you replace the event?

RixBai, 2020-01-31:
No, I didn't solve the issue yet.

sibelius, 2020-01-31:
or we rename our event name or they need to rename theirs

RixBai, 2020-02-10:
How to rename our event name, @sibelius ?

yonghongren, 2020-02-20:
react-native-document-scanner also defines PictureTaken event. I get around this by renaming PictureTaken in react-native-document-scanner/ios/*.[h,m] to ScanTaken.

closed 2020-04-27 by stale[bot]

divyanshunegi, 2020-05-14:
what is the solution?

njacob1001, 2020-06-22:
what about the solution?

JP6720, 2020-07-10:
same issue here

davidvmckay, 2021-03-10:
@divyanshunegi  @njacob1001 @JP6720 

Following suggestion of @cristianoccazinsp, I went in to my Pods project, /Development Pods/RNPdfScanner and replaced all textual instances of `onPictureTaken` in its .h and .m files with `onDocumentScanned`, and also updated the event name in my TypeScript code.

I think I'll need to remember to fix it every time I `pod update` or `pod install`, so I also put jsdocs linking to this GitHub issue on my type declarations for the module.

> A similar issue was reported here: [react-native-video/react-native-video#1625](https://github.com/react-native-video/react-native-video/pull/1625)
> 
> Their fix was to have the library replace the event type. You might have luck replacing the event on https://github.com/Michaelvilleneuve/react-native-document-scanner/blob/master/ios/DocumentScannerView.h#L6 , but no idea if it would still work. If anything, react-native-document-scanner would probably benefit by replacing the event with the direct event type instead.

RixBai, 2021-04-25:
Is there any updates?

vignesh4622, 2022-08-17:
any solution?