react-native-camera/react-native-camera #2621: Typo in mirrorVideo => Exported url is not changed to .mp4

MordiukRoman, 2019-12-03:
# Bug Report

**To Do First**

- [+] Did you try latest release?
- [+] Did you try master?
- [-] Did you look for existing matching issues?

**Platforms**
iOS

**Description/Current Behaviour**

line 1840: - NSURL* outputUR

**Expected Behaviour**

should be: - NSURL* outputURL

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-12-03:
That's a nice catch, what problem does it cause exactly? I've seen mirrorVideo work fine on iOS.

Are you sure that typo wasn't intended? The "right" variable is defined below 
https://github.com/react-native-community/react-native-camera/blob/1f8b86394747e116f4936b6cb408ad8a5b78d85c/ios/RN/RNCamera.m#L1866

Since it is just a callback/handler, I don't think the name typo really causes any issues, but I may be wrong.

MordiukRoman, 2019-12-03:
It causes file extension to be default like ".mov".
After I fixed the typo video exported in ".mp4".

cristianoccazinsp, 2019-12-03:
Interesting. I guess I didn't see that one since I'm always saving files as ".mov" because I use a custom path to store the videos. Good catch, you should provide a PR :)

MordiukRoman, 2019-12-03:
`(void)mirrorVideo:(NSURL *)inputURL completion:(void (^)(NSURL* outputURL))completion {
    AVAsset* videoAsset = [AVAsset assetWithURL:inputURL];
    AVAssetTrack* clipVideoTrack = [[videoAsset tracksWithMediaType:AVMediaTypeVideo] firstObject];

    AVMutableComposition* composition = [[AVMutableComposition alloc] init];
    [composition addMutableTrackWithMediaType:AVMediaTypeVideo preferredTrackID:kCMPersistentTrackID_Invalid];

    AVMutableVideoComposition* videoComposition = [[AVMutableVideoComposition alloc] init];
    videoComposition.renderSize = CGSizeMake(clipVideoTrack.naturalSize.height, clipVideoTrack.naturalSize.height);
    videoComposition.frameDuration = CMTimeMake(1, 30);

    AVMutableVideoCompositionLayerInstruction* transformer = [AVMutableVideoCompositionLayerInstruction videoCompositionLayerInstructionWithAssetTrack:clipVideoTrack];

    AVMutableVideoCompositionInstruction* instruction = [[AVMutableVideoCompositionInstruction alloc] init];
    instruction.timeRange = CMTimeRangeMake(kCMTimeZero, CMTimeMakeWithSeconds(60, 30));

    CGAffineTransform transform = CGAffineTransformMakeRotation(M_PI/2.0);
    transform = CGAffineTransformTranslate(transform, 0.0, -clipVideoTrack.naturalSize.height);

    [transformer setTransform:transform atTime:kCMTimeZero];

    [instruction setLayerInstructions:@[transformer]];
    [videoComposition setInstructions:@[instruction]];`

closed 2020-02-08 by stale[bot]