react-native-camera/react-native-camera #2624: Can't get full text of QR Code

dudgns0908, 2019-12-03:
# Bug Report

**To Do First**

- [ ] Did you try latest release?  Yes
- [ ] Did you try master?  Yes
- [ ] Did you look for existing matching issues?  Yes

**Platforms**

**Versions**
- Android: all
- iOS: all
- react-native-camera: 3.13.1
- react-native: 0.61.2
- react: 16.9.0

when read qrcode, I can not get full text that below text
> 911,,,,9先　ABC医院　0000-11-2222ください。,1」,2,錠,2,62222090

But This QR Code has this text:

> JAHISTC04
> 1,サンプル　太郎,1,19691010,,,,,,
> 2,,,9
> 3,,,,9
> 4,,,9
> 5,20170416
> 11,ABC医院,20,1,1234567,,,
> 15,サンプル　花子,0000-11-2222,1
> 51,ABC医院,20,1,123456755,石田　公孝,
> 201,1,ネキシウムカプセル２０ｍｇ,1,カプセル,2,622080801
> 281,1,,1,
> 291,1,,1,
> 301,1,朝食後,35,日分,1,1,
> 311,1,,1,
> 391,1,,1,
> 201,2,アムロジピン錠５ｍｇ「ツルハラ」,1,錠,2,620007877
> 281,2,,1,
> 291,2,グレープフルーツジュースは飲まないようにしましょう,1,
> 301,2,朝食後,35,日分,1,1,
> 311,2,,1,
> 391,2,車の運転に注意が必要です,1,
> 201,3,アトルバスタチン錠１０ｍｇ「サンド」,1,錠,2,622127001
> 281,3,,1,
> 291,3,グレープフルーツジュースは飲まないようにしましょう,1,
> 301,3,朝食後,35,日分,1,1,
> 311,3,,1,
> 391,3,,1,
> 201,4,ラックビー錠,2,錠,2,622113001
> 281,4,,1,
> 291,4,,1,
> 301,4,朝夕食後,35,日分,1,1,
> 311,4,,1,
> 391,4,,1,
> 201,5,フェキソフェナジン塩酸塩錠６０ｍｇ「ＥＥ」,2,錠,2,622220901
> 281,5,,1,
> 291,5,,1,
> 301,5,朝夕食後,35,日分,1,1,
> 311,5,,1,
> 391,5,,1,
> 401,他の薬を併用する際は、相談してください。,1
> 411,,9
> 501,連絡先　ABC医院　0000-11-2222
> 601,,
> 701,,,,9
> 911,,,

dudgns0908, 2019-12-03:
@sibelius IOS also has this problem

sibelius, 2019-12-03:
can you dig in native code to find the issue?

dudgns0908, 2019-12-03:
I have no idea...
how to access native code?

cristianoccazinsp, 2019-12-04:
Looks like a very convoluted QR Code! Have you tried other reads to see if they have the same issue?

dudgns0908, 2019-12-04:
@cristianoccazinsp IPhone and Android's general camera's Scanner can read it. however, RNCamera's onBarCodeRead function can not read it.

dudgns0908, 2019-12-04:
[Solved] That's QR code doesnt have  '\n', It is only has '/r'.

closed 2019-12-04 by sibelius