react-native-camera/react-native-camera #2626: Metro bundler encountered an error while trying to resolve module

XamarinPOC, 2019-12-03:
# Bug Report
Error: InternalError Metro Bundler has encountered an error: While trying to resolve module `react-native-camera` from file `/Users/akumar/Documents/ReactPractice/ImageSample/App.js`, the package `/Users/akumar/Documents/ReactPractice/ImageSample/node_modules/react-native-camera/package.json` was successfully found. However, this package itself specifies a `main` module field that could not be resolved (`/Users/akumar/Documents/ReactPractice/ImageSample/node_modules/react-native-camera/src/index.js`. Indeed, none of these files exist:

  * /Users/akumar/Documents/ReactPractice/ImageSample/node_modules/react-native-camera/src/index.js(.native|.ios.js|.native.js|.js|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)
  * /Users/akumar/Documents/ReactPractice/ImageSample/node_modules/react-native-camera/src/index.js/index(.native|.ios.js|.native.js|.js|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)

showCompileError
    index.bundle?platform=ios&dev=true&minify=false:34620:26
<unknown>
    index.bundle?platform=ios&dev=true&minify=false:34559:29
emit
    index.bundle?platform=ios&dev=true&minify=false:35039:35
<unknown>
    index.bundle?platform=ios&dev=true&minify=false:34886:23
dispatchEvent
    index.bundle?platform=ios&dev=true&minify=false:32357:31
<unknown>
    index.bundle?platform=ios&dev=true&minify=false:32991:31
emit
    index.bundle?platform=ios&dev=true&minify=false:6944:42
__callFunction
    index.bundle?platform=ios&dev=true&minify=false:2695:49
<unknown>
    index.bundle?platform=ios&dev=true&minify=false:2408:31
__guard
    index.bundle?platform=ios&dev=true&minify=false:2649:15
callFunctionReturnFlushedQueue
    index.bundle?platform=ios&dev=true&minify=false:2407:21
callFunctionReturnFlushedQueue
    [native code]:0

**Platforms**

iOS

**Versions**

{
  "name": "ImageSample",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "react": "16.9.0",
    "react-native": "0.61.5",
    "react-native-camera": "^3.13.1",
    "save": "^2.4.0"
  },
  "devDependencies": {
    "@babel/core": "7.7.4",
    "@babel/runtime": "7.7.4",
    "@react-native-community/eslint-config": "0.0.5",
    "babel-jest": "24.9.0",
    "eslint": "6.7.2",
    "jest": "24.9.0",
    "metro-react-native-babel-preset": "0.56.3",
    "react-test-renderer": "16.9.0"
  },
  "jest": {
    "preset": "react-native"
  }
}

cristianoccazinsp, 2019-12-04:
Have you checked if the file is in fact there? This normally happens when installing/updating libraries. Just close all your metro bundlers / consoles and start fresh.

hamza-ameer, 2019-12-20:
@cristianoccazinsp  That's worked for me. thanks dude.

closed 2019-12-20 by sibelius

rocky1973, 2020-02-24:
Thanks mate, that sorted my issues.

deionstfleur, 2021-01-08:
I am still getting this issue, anybody have any idea how I can fix this?

eldirb21, 2021-03-04:
I am still getting this issue, too

sountstars, 2021-05-10:
I am still getting this issue, too

Gosmacx, 2022-03-11:
m2

Vaib19121, 2022-03-12:
m2

aafaqahmed, 2022-07-03:
> Have you checked if the file is in fact there? This normally happens when installing/updating libraries. Just close all your metro bundlers / consoles and start fresh.

Appreciated!
This resolved my error!