react-native-camera/react-native-camera #2628: Not detecting any faces on iOS

viimauer, 2019-12-05:
# Bug Report

**To Do First**

- [YES] Did you try latest release?
- [YES] Did you try master?
- [YES] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 12.4.3
- react-native-camera: 3.13.1
- react-native: 0.61.3
- react: 16.9.0
  

**Description/Current Behaviour**

onFacesDetected is not detecting any faces on back camera, just return empty array. In from camera it works as expected.

**Expected Behaviour**

Detect faces when there is faces in the camera and return a filled array

**Steps to Reproduce**

Install react-native-camera via cocoa pods

pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
'FaceDetectorMLKit'
]

**Additionals**

Code example:

```
        <RNCamera
            androidCameraPermissionOptions={androidCameraOptions}
            captureAudio={false}
            faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all} // tried all options
            faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.accurate} // tried all options
            flashMode={RNCamera.Constants.FlashMode.off}
            onFacesDetected={this.onFaceDetected}
            ref={ref => (this.camera = ref)}
            style={styles.container}
            type={RNCamera.Constants.Type.back}>
        ...
        </RNCamera>
```

also tried all defaultVideoQuality options

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sireesha-9, 2019-12-28:
I am also facing the same issue. Any resolution?

sireesha-9, 2019-12-28:
In my scenario, *onFacesDetected* is not triggering in front camera and *RNCamera.Constants.FaceDetection* values are coming nil. Please help me out in this. Thank you

kyle-johnson, 2020-01-20:
This is might be an MLKit issue: https://github.com/firebase/quickstart-ios/issues/874

kettyy, 2020-02-21:
I am also facing the same issue. Any resolution?

kettyy, 2020-04-29:
@viimauer Did you solve it?

tomkos, 2020-05-20:
Any one has some work around to it?

luvnish19, 2020-07-02:
@heraldovalenti 
I am using camera not mlkit and tried with your both ways but still "onFacesDetected" method is not called in iOS. In android is working without any doubt.

If you help me then I will be very thankful to you.

My source code:
<RNCamera
                    ref={ref => { this.camera = ref; }}
                    style={{
                        width: '100%',
                        height: '100%',
                        justifyContent: 'space-between',
                    }}
                    zoom={this.state.zoom}
                    maxZoom={MAX_ZOOM}
                    type={this.state.cameraType}
                    flashMode={flash}
                    autoFocus={this.state.autoFocus}
                    whiteBalance={this.state.whiteBalance}
                    onFacesDetected={this.state.faceDetectionEnabled ? this.facesDetected : undefined}
                    onFaceDetectionError={() => console.log("this.faceDetectedError")}
                    ratio={this.state.ratio}
                    focusDepth={this.state.depth}
                    onCameraReady={() => this.setState({ faceDetectionEnabled: true })}
                    captureAudio={false}
                    androidCameraPermissionOptions={{
                        title: 'Permission to use camera',
                        message: 'We need your permission to use your camera',
                        buttonPositive: 'Ok',
                        buttonNegative: 'Cancel',
                    }} >

                </RNCamera>

heraldovalenti, 2020-07-06:
Hey @luvnish19 

I've reviewed your solution and seems ok, it has the same logic I've used to workaround this issue.

Have you tried the basic example for **mlkit**? I mean, the one available on 
[examples/mlkit](https://github.com/react-native-community/react-native-camera/tree/master/examples/mlkit)

Would you try to execute the scenario I've described, to verify if you get the same result as me?:
1. Run the example
2. Enable face scan > onFacesDetected is never called
3. Disable it
4. Enable face scan again > onFacesDetected is now called

If you get the same result, then we would discard any device OCR issue related, and we can continue from there.
Thanks!

locliem08, 2020-07-22:
I fixed detection on back camera IOS: [](https://github.com/react-native-community/react-native-camera/issues/1870#issuecomment-662305158)

closed 2020-09-27 by stale[bot]