react-native-camera/react-native-camera #2629: BarCodeType org.gs1.EAN-13

smeevil, 2019-12-05:
Hi,

I'm using react-native-camera to use barcode scanning with the build in `onBarCodeRead` callback. The response I get according to the types (index.d.ts) is an object that contains `data`, `rawData`, `type` and `bounds`. 

I have two questions about this.

- The type that I'm getting from a scan is `org.gs1.EAN-13` which is not included in the `BarCodeType` and makes typescript unhappy if I'm trying to match on that. Is there something that I'm doing wrong since it seems that the expected type should be `ean13`

- I'm also getting a field with `target` that contains a number (427), but I can't find the purpose of this in the documentation. 

I'm using the component as follows.

```JSX
<RNCamera
        style={styles.preview}
        type={RNCamera.Constants.Type.back}
        captureAudio={false}
        flashMode="torch"
        onBarCodeRead={parseBarCode}
>
 ```

```typescript
import { BarCodeType } from 'react-native-camera/types'

type TBarCodeData = {
  data: string
  rawData?: string
  type: keyof BarCodeType
}

const parseBarCode = (data: TBarCodeData): Promise<void> => {...}
```

Thanks for your time :)
Gerard.

P.S.: This is on iOS (13.2)

gigeos, 2020-01-06:
I have the same issue on iOS 12.4 :(

IrmantasVaracinskas, 2020-03-02:
Real barcode type is just `BarCodeType` instead of `typeOf BarCodeType`. Also to match on that type try using  `RNCamera.Constants.BarCodeType.ean13`

closed 2020-05-09 by stale[bot]