react-native-camera/react-native-camera #2632: TypeError: undefined is not an object ( evaluating '_reactNativeCamera.RNCamera.constants.Aspect.fill')

nihp, 2019-12-09:
Getting this error TypeError: undefined is not an object ( evaluating '_reactNativeCamera.RNCamera.constants.Aspect.fill')

cristianoccazinsp, 2019-12-09:
Don't know about the first error, but for the second error, are you getting as a warning, or as a device crash? Looks to me it's just a warning because you cannot record audio on the device you're testing (probably the simulator?). If it's a warning, it's probably safe to ignore unless you are having issues recording with audio. You may also try with `captureAudio={false}`

nihp, 2019-12-16:
I have added captureAudio={false} 

This is my code. I can't able to take capture. Can you check this.

Getting error undefined is not an object evaluating this.camera.takePictureAsync

```
takePicture = async function() {
    // alert('1111')
          const options = { quality: 0.5, base64: true,  };
          const data = await this.camera.takePictureAsync(options);
           this.setState({path: data.uri})
           ImagePicker.openCropper({
             path: data.path,
             width: deviceWidth,
             height: deviceWidth*5/4,
             cropperToolbarTitle: '',
             hideBottomControls: true,
             enableRotationGesture: true
           }).then(image => {
             this.props.screenProps.navigation.navigate(this.props.screenProps.navigation.state.params.screenName, {uri: image.path})
           })
           .catch((err) => {
             console.log("openCropper error = " + err)
           }).catch((error) => {
           console.log(error)
         })
     }

<View>
        <RNCamera
          style={styles.view}
          ref={cam => {
            this.camera = cam;
          }}
          type={RNCamera.Constants.Type.back}
        //  style={styles.preview}
          //aspect={RNCamera.Constants.Aspect.fill}
          captureAudio={false}
          >
        </RNCamera>
      </View>
        <Icon style = {{ alignSelf:'center',}}onPress={this.takePicture}  name="camera" size={50}/>
  </View>

```

cristianoccazinsp, 2019-12-16:
`undefined is not an object evaluating this.camera.takePictureAsync`  tells me that your ref hasn't even been set? Is the camera even mounting? Does it show anything? Really looks like a programming error.

Try adding these for debugging as well:
```
onStatusChange={(s) => { console.log(s) }}
onCameraReady={() => {console.log("ready") }}
onMountError={() => console.log("mount error") }}
```

nihp, 2019-12-17:
Worked for me. I need to test that in real devices and let you know

cristianoccazinsp, 2019-12-17:
What worked exactly? What was the issue?

nihp, 2019-12-17:
I have changed my options to empty and then changed the image path in
imagecropper. It works fine now

closed 2020-02-22 by stale[bot]