react-native-camera/react-native-camera #2634: not work on android 4.2.2

moh76, 2019-12-09:
hi 
 <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={{
            flex: 1,
            width: '100%',
          }}
          onGoogleVisionBarcodesDetected={this.barcodeRecognized}
   > </RNCamera> 
        

this code worked on android v7 but on v4.2.2 show below error

"error while updating property 'cameraId' of a view managed by: RNCamera"

how fix it ?

cristianoccazinsp, 2019-12-09:
Can you try without barcodes first? Can you get an error stack trace from android studio / logcat?

moh76, 2019-12-10:
i tried #2635 and it worked 

thank you 

> Can you try without barcodes first? Can you get an error stack trace from android studio / logcat?
> 
> Related issue: #2528
> 
> Not sure why it was closed, did we ever get a PR to fix it? In short, this will crash with Android below API level 19 (Android 4.4).
> 
> Update: Can you try with this PR? #2635

closed 2020-02-15 by stale[bot]