react-native-camera/react-native-camera #2637: takePictureAsync throwing an error "image could not be captured"

elirangoshen, 2019-12-10:
# Bug Report

**To Do First**

- [v] Did you try latest release?
- [v] Did you try master?
- [v] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS: 13.2.2
- react-native-camera: 3.14.0
- react-native: 0.61.2
- react: 16.9.0
  

**Description/Current Behaviour**

When i take a picture with my device using takePictureAsync im getting an error : 
Image  could not be captured
my device is iphone 8 ios 13.2.2

**Expected Behaviour**

that should capture the photo

**Steps to Reproduce**

im not using expo

**Additionals**

this is my RNCamera component : 

`        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.auto}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          captureAudio={false}
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
        />`

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-12-10:
Is the camera preview running? Any chance you can get the full error logs from xcode?

elirangoshen, 2019-12-11:
> Is the camera preview running? Any chance you can get the full error logs from xcode or the additional error info?

how i get the additional error info ? 
I'm getting it in android as well..  I don't know what changed as its a version that worked before.

elirangoshen, 2019-12-17:
? can you try it with firebase libs. I found ther solution but it still strange why it doesnt work

vwpo, 2019-12-27:
My fault with this error was when I was trying perform takePictureAsync and remove RNCamera element from DOM to display loader. Make sure you doesn't removing RNCamera from DOM when capturing

closed 2020-03-03 by stale[bot]

pooja-wal, 2021-05-31:
@vwpo , thank you. I was running into the same issue only on iOS and your answer is really helpful.

longshanzhiyu, 2021-12-29:
> @vwpo , thank you. I was running into the same issue only on iOS and your answer is really helpful.

Found the reason for this problem?How to solve it?

tuanthai-agilityio, 2022-02-10:
@longshanzhiyu I'm having the same issue, did you solve it?