react-native-camera/react-native-camera #2638: Android - pictures always upside down with locked orientation no matter what the "orientation" setting is

allthetime, 2019-12-11:
**Versions**
- react-native-camera: 3.8.0
- react-native: 0.60.4
- react: 16.8.6

I have an app where you have to take some photos with the phone in a specific orientation. The user is guided to do so and so I can guarantee how the phone will be held when the picture is taken. 

On iOS I am checking the phone orientation and then setting the `orientation` setting on `takePictureAsync` accordingly. This reliably leads to good results on all devices. 

The same method does not work for android at all. 

Because I know what way the user will be holding the phone already I have attempted to hard code the `orientation` values into `takePictureAsync` but it doesn't seem to have an effect. Infact, regardless of whether I send orientation = `landscapeRight` or `landscapeLeft` the pictures still come in upside down. 

The return data is different though, with `landscapeRight` I get:

`{ pictureOrientation: 4, deviceOrientation: 4 }`

and with `landscapeLeft` set I get:

`{ pictureOrientation: 3, deviceOrientation: 4 }`

Even though the results are different in data, the picture is still upside down.

How can I get RNCamera to respect the `orientation` setting? `fixOrientation` also has no effect.

cristianoccazinsp, 2019-12-11:
Can you try with the latest RNCamera version? Have you tried multiple devices? I do seem to get photos in the right orientation on my 2 android devices, even with orientation locked. I'm not using camera2 api also.

pontusab, 2019-12-11:
I have the same issue with the latest RNCamera and latest RN.

sibelius, 2019-12-11:
android or ios?

cristianoccazinsp, 2019-12-11:
Can you guys post a sample component, device and OS version, and RNCamera version?

Just to be sure, this happens with locked UI orientation. I haven't been able to reproduce it on Android nor iOS.

allthetime, 2019-12-11:
@cristianoccazinsp 

I'm getting ahold of a phone tomorrow that behaves differently than my own, I'm going to test both under a few different conditions I'll post the code and results in a couple days.

cristianoccazinsp, 2019-12-17:
Related: https://github.com/react-native-community/react-native-camera/issues/2645

closed 2020-02-22 by stale[bot]