react-native-camera/react-native-camera #2639: Crash on iOS

ivela, 2019-12-11:
I'm having a crash on iOS frequently when users try to take a picture with RNCamera

This is the code that I'm using to take the picture:

takePicture = async function() {
    let registerPictureWidth = global.registerPictureWidth;
    
    if (this.camera) { 
      const options = { quality: 0.5, base64: true, width: registerPictureWidth  };
      const data = await this.camera.takePictureAsync(options);
      this.props.navigation.goBack();
      this.props.navigation.state.params.onSelect({ pictureUri: data.uri });
    }
  }; 

This is the crash stack given by Fabric.

```
Fatal Exception: NSInvalidArgumentException
0  CoreFoundation                 0x1ce9bb3d __exceptionPreprocess
1  libobjc.A.dylib                0x1c123067 objc_exception_throw
2  AVFoundation                   0x237d2a2d -[AVCaptureStillImageOutput captureStillImageAsynchronouslyFromConnection:completionHandler:]
3  iDoctus                        0x5a41c7 -[RNCamera takePicture:resolve:reject:] + 381 (RNCamera.m:381)
4  iDoctus                        0x5a4051 __54-[RNCamera takePictureWithOrientation:resolve:reject:]_block_invoke + 368 (RNCamera.m:368)
5  iDoctus                        0x5afaef __60-[RNSensorOrientationChecker getDeviceOrientationWithBlock:]_block_invoke + 66 (RNSensorOrientationChecker.m:66)
6  iDoctus                        0x5af939 __36-[RNSensorOrientationChecker resume]_block_invoke + 49 (RNSensorOrientationChecker.m:49)
7  Foundation                     0x1d7c1b01 (Missing)
8  Foundation                     0x1d717aef (Missing)
9  Foundation                     0x1d709fe7 (Missing)
10 Foundation                     0x1d7c3d39 (Missing)
11 libdispatch.dylib              0x1c57659d _dispatch_queue_serial_drain
12 libdispatch.dylib              0x1c56cb71 _dispatch_queue_invoke
13 libdispatch.dylib              0x1c5781b5 _dispatch_root_queue_drain
14 libdispatch.dylib              0x1c57800f _dispatch_worker_thread3
15 libsystem_pthread.dylib        0x1c71e87d _pthread_wqthread
16 libsystem_pthread.dylib        0x1c71e45c start_wqthread
```

cristianoccazinsp, 2019-12-11:
What version of RNCamera? I see the `Fatal Exception: NSInvalidArgumentException`, but I can't tell precisely what's wrong. Looks like the capture call is receiving some unexpected parameter. 

Would it be possible for you to debug with xcode a bit more? Have you tried without navigating away? Any chance you are calling this method before the camera is ready (try listening to `onStatusChange` to get the ready event)?

BatDev0, 2019-12-25:
What version of RNCamera? I see the ```Fatal Exception: NSInvalidArgumentException```, but I can't tell precisely what's wrong. Looks like the capture call is receiving some unexpected parameter.

Would it be possible for you to debug with xcode a bit more? Have you tried without navigating away? Any chance you are calling this method before the camera is ready (try listening to onStatusChange to get the ready event)?

It's also odd that that whole code is within a try/catch, so it should at least not crash the app.

closed 2020-03-01 by stale[bot]