react-native-camera/react-native-camera #2640: Unable to Scan this EAN-13

Pczek, 2019-12-12:
**Platforms**
iOS

**Versions**

- Android:
- iOS: 13.2.3
- react-native-camera: 3.13.1
- react-native: 0.60.5
- react: 16.9.0

**Description/Current Behaviour**
I'm using the `onBarCodeRead` API on iOS and it works great so far. However, I can't read this specific code. Can someone tell me why? I'm using the Google MLKit lib on Android and it works fine.

This is my barcode types setup. This is also a ean13 code right?
```
barCodeTypes: [RNCamera.Constants.BarCodeType.ean13],
```

[image]

**Expected Behaviour**
I it should read the barcode.

**Steps to Reproduce**

Try to scan the barcode screenshot

cristianoccazinsp, 2019-12-12:
I cannot seem to scan that code neither; however, I can scan EAN13 codes from google just fine. Any chance it is a bad code?

Pczek, 2019-12-13:
When I generate a EAN-13 Code with the numbers it works fine. The generator also checks if the code is valid and otherwise throws an error that the check digit is wrong.

[image]

cristianoccazinsp, 2019-12-13:
> When I generate a EAN-13 Code with the numbers it works fine. The generator also checks if the code is valid and otherwise throws an error that the check digit is wrong.
> 
> [image]

That code works fine for me on both platforms. The issue is the same code without the numbers?

Pczek, 2019-12-15:
The generated code without the numbers works fine for me as well. I can't make out why the photo of the original code can't be read by the library used by RNC. Can't see any distinct issue with the photo i've posted in the opening of this issue.

Pczek, 2019-12-20:
I've made this comparison to check if the pattern's differ. However, they are exactly the same. See below. Nevertheless, there are some parts which are not perfectly printed or photographed. Not sure what's causing the read to fail though, would be bad if the recognition fails due to such minor disturbances.

[image]

[image]

Pczek, 2019-12-20:
Using `3.15.0` as of now. I was able to read the code a few times however it was still a hassle, but better than before. Don't know why. :shrug:

cristianoccazinsp, 2019-12-20:
Is the issue something you can consistently make it happen? Maybe it was just that one code? Or the camera you're using is not capturing properly (the one you use to take the photo of the QR, not to read it). I don't think the reading logic is even implemented by the library, that's just google/apple code.

closed 2020-02-25 by stale[bot]