react-native-camera/react-native-camera #2641: Failed to resolve: com.google.android.gms:play-services-vision

alex-mironov, 2019-12-13:
I'm using react-native-camera `v.3.11.1`. On an attempt to build the project I'm getting: ```
ERROR: Failed to resolve: com.google.android.gms:play-services-vision:16.1.0
```

`android/build.gradle`
```
buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 28
        googlePlayServicesVersion = "16.1.0" // default: "+"
        firebaseVersion = "17.3.4" // default: "+"
    }
...
```

Any suggestion what could be wrong?

OkancanCosar, 2019-12-13:
not sure is related but im using computer language setting turkish.
when i get this error, like this;

 > A problem occurred evaluating project ':react-native-camera'.
   Could not find method generalImplementation() for arguments [com.google.android.gms:play-services-vision:17.0.2] on object of 
  type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler
  
change this file
  > node_modules\react-native-camera\android\build.gradle

   > on dependencies/ node

```diff
- generalImplementation 
+ generalİmplementation
```
and
```diff
- mlkitImplementation   
+ mlkitİmplementation
```

thats ridiculous but work,
btw u can check react-native-camera setup instructions that i write on  [....Gist....](https://gist.github.com/OkancanCosar/a7590c51275261b1515fb0585240484f)

cristianoccazinsp, 2019-12-13:
Any chance this is a docs issue? I actually haven't seen version 16.+ referenced on the docs at all.

I've been using `17.+` for a long time now (I believe this was changed when Androidx support was added). Try using it.

zhenliliu, 2019-12-15:
I have face the same problem, have you fixed this isseu?

xchwarze, 2020-01-21:
This fixed the problem for me

```
buildscript {
    ext {
        ...
        googlePlayServicesVersion = "17.0.2"
    }
```

error:
`Could not find com.google.android.gms:play-services-vision:16.0.0.`

buiquangduc, 2020-02-01:
Same here, did you resolve this issue? @alex-mironov

abhaynpai, 2020-02-06:
If you check the following line, you can override the `googlePlayServicesVisionVersion` in your `android/build.gradle` file.

https://github.com/react-native-community/react-native-camera/blob/c683076a4810a8a0fec490683fb3c4a62d123868/android/build.gradle#L79

All you need to is the following -
```
buildscript {
    ext {
        googlePlayServicesVisionVersion = "19.0.0"
    }
}
```

This worked for me.

closed 2020-04-13 by stale[bot]

vinhnglx, 2020-06-17:
I tried this way (https://github.com/react-native-community/react-native-camera/issues/2414#issuecomment-524971175) and it works

my `android/build.gradle`

```
ext {
    minSdkVersion = 26
    googlePlayServicesVersion = "16.1.0"
    googlePlayServicesVisionVersion = "16.2.0"
    buildToolsVersion = "28.0.3"
    compileSdkVersion = 28
    targetSdkVersion = 28
    supportLibVersion = "28.0.0"
    firebaseIidVersion = "20.2.0"
}
```