react-native-camera/react-native-camera #2645: Issue with users taking landscape pictures

enigmablue, 2019-12-16:
# Question

I can't seem to figure out how to support users who want to take pictures on landscape mode.

As i am locking my app on portrait mode, the images come out vertical even when the picture is taken horizontal. And when the app is locked on portrait, Dimensions don't really work.

How are some of you handling users who are taking landscape pictures?

cristianoccazinsp, 2019-12-16:
Android or iOS? I'm sure I've been taking pictures with the phone rotation locked and they always come up with the right orientation with the latest RNCamera version.

If Android, are you using Camera1 or Camera2? What device? Got some sample code?

enigmablue, 2019-12-17:
Hi @cristianoccazinsp ! Thanks for responding.

Both android and ios are having the same issue. My react native camera version is:
  "react-native-camera": "^3.13.0",

It appears on all kinds of devices i believe, on my samsung, or some chinese brands (on android go) and also my user's phones. Not sure about Camera1 or Camera2 ...

`  takePicture = async() => {
    if (this.camera) {
      let data = await this.camera.takePictureAsync({
        quality: 1,
        base64: true,
        pauseAfterCapture: true,
        forceUpOrientation: true,
        fixOrientation: true,
        orientation: "portrait"
      });
      delete data.base64;
      this.props.navigation.goBack();
      data = {...data,type : "image"}

      this.props.navigation.state.params.cameraResponse(data);
    }
  };`

My RN Camera instantiation is

`  <RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            skipProcessing={true}
            style={styles.preview}
            type={this.state.cameraType}
            flashMode={this.state.flashMode}
            // captureQuality={'480p'}
            defaultVideoQuality={VIDEO_QUALITY}
            androidCameraPermissionOptions={{
              title: 'Permission to use camera',
              message: 'We need your permission to use your camera',
              buttonPositive: 'Ok',
              buttonNegative: 'Cancel',
            }}
            androidRecordAudioPermissionOptions={{
              title: 'Permission to use audio recording',
              message: 'We need your permission to use your audio',
              buttonPositive: 'Ok',
              buttonNegative: 'Cancel',
            }}>
</RNCamera>
`

cristianoccazinsp, 2019-12-17:
Can you try removing some of the capture options one by one? Remove
orientation, then the fix orientation flags. Lastly, try without base64.
See if any of those changes get you the right output. I'm quite skeptical
about the orientation prop. The fix orientation flag should be enough

enigmablue, 2019-12-17:
> Can you try removing some of the capture options one by one? Remove orientation, then the fix orientation flags. Lastly, try without base64. See if any of those changes get you the right output. I'm [...]

Will give it a shot thank you!

enigmablue, 2019-12-17:
Removing those other orientation flags did fix the funny orientation. 
But now it dosen't take the full length of the picture, only the middle part of the landscape, and then shows it on portrait. 

On the regular phone camera and gallery it shows like this, [https://i.imgur.com/OhBki5n.png](url)
Can it be achieved?

cristianoccazinsp, 2019-12-17:
Not entirely sure what's the issue. I have been using the camera with
locked rotation just fine and the picture always comes up with the right
orientation. But looks like some of those flags are messing up the picture

enigmablue, 2019-12-17:
FIXED! Thanks i had to update my resize mode (on fastimage) to contain so the whole picture can be seen

resizeMode={FastImage.resizeMode.contain}

closed 2019-12-17 by enigmablue

cristianoccazinsp, 2019-12-17:
So, which one of all the props/flags was causing the picture to end up with the wrong rotation?

enigmablue, 2019-12-17:
This is my final result - also tested on android. I basically removed all the orientations.
On latest camera

```
  takePicture = async() => {
    if (this.camera) {
      let data = await this.camera.takePictureAsync({
        quality: 1,
        base64: true,
        pauseAfterCapture: true,
  //      forceUpOrientation: true,
  //      fixOrientation: true,
//        orientation: "landscapeLeft"
      });
      delete data.base64;
      this.props.navigation.goBack();
      data = {...data,type : "image"}

      this.props.navigation.state.params.cameraResponse(data);
    }
  };
```

cristianoccazinsp, 2019-12-17:
If I had to guess, `orientation: "landscapeLeft"` was breaking it.  `forceUpOrientation` and `fixOrientation`  would basically replace EXIF rotation with real rotation, and those two are also the settings I'm using and haven't had issues.

It would be good to confirm that the `orientation` prop is in fact breaking the image (honestly, I'm not sure why that prop even exists).