react-native-camera/react-native-camera #2647: Remove ActionSheet asking to take a photo or to choose from library

biancaFar, 2019-12-18:
# Question

**To Do First**
- [x ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I have an app, where I need to open the camera and take a picture when a custom button is pressed. However, each time the camera opens, I get the ActionSheet as displayed in the attached image. My question is, how can I remove that ActionSheet? 

I event tried to use the code from one of the examples provided by
 the library, but the ActionSheet is still displayed.

[image]

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2019-12-19:
Are you sure this is from RNCamera?  Can you try the other sample? Can you show some code? What RNCamera version?

biancaFar, 2019-12-19:
I use RN Camera 3.8.0 and RN version 0.60.0. The thing is we updated RN version from 0.50 to 0.60 and it used an older react-native camera version, but the pop-up starting to show only after upgrade. Regarding the code, I pretty much copy-pasted the code from the sample in the whole screen. There is nothing modified.

cristianoccazinsp, 2019-12-19:
I would suggest upgrading RNCamera first. But I've never seen this, nor why ActionSheet would be related to RNCamera at all.

You also did a massive jump from RN 50 to 60, so might want to re view all that.

sibelius, 2019-12-19:
share a repro

we don't have any actionsheet in the core library

closed 2019-12-19 by sibelius