react-native-camera/react-native-camera #2661: Crash if app is minimized in the moment after video recording has started

md-cherepanov, 2019-12-30:
# Bug Report

**To Do First**

- [✓] Did you try latest release?
- [✓] Did you try master?
- [✓] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 7/8/9
- react-native-camera: 3.15.0
- react-native: 0.61.4
- react: 16.9.0

**Description/Current Behaviour**

If the app is minimized right after video recording has started, then crash happens in native code causing app to exit.

**Expected Behaviour**

App doesn't crash. Video recording process is either continues or interrupts, but without any errors in native code.

**Steps to Reproduce**

1. Navigate to screen with mounted RNCamera component.
2. Invoke recordAsync.
3. Immediately (in one second or less) minimize the app (tap Home or Overview button).

**Additionals**

Notes

Tried to stop recording manually or unmount camera component on app minimizing, but crash still continued to occur in such cases.

If the video recording flow takes significant part of the app and because we have no control over user minimizing the app in any moment, such crash leads to bad UX and has a huge impact on app usability.

Usage example

Mount:
`<RNCamera captureAudio={false} ref={this.camera} style={{ flex: 1 }} />`

Start recording:
`await this.camera.current.recordAsync();`

Crash log

<pre>
2019-12-30 11:31:22.701 6821-6821/com.x.x E/MediaRecorder: stop called in an invalid state: 8
2019-12-30 11:31:22.770 6821-6821/com.x.x E/CAMERA_1::: mMediaRecorder.stop() failed
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.google.android.cameraview.Camera1.stop(Camera1.java:214)
        at com.google.android.cameraview.CameraView.stop(CameraView.java:334)
        at org.reactnative.camera.RNCameraView.onHostPause(RNCameraView.java:512)
        at com.facebook.react.bridge.ReactContext.onHostPause(ReactContext.java:240)
        at com.facebook.react.ReactInstanceManager.moveToBeforeResumeLifecycleState(ReactInstanceManager.java:673)
        at com.facebook.react.ReactInstanceManager.onHostPause(ReactInstanceManager.java:497)
        at com.facebook.react.ReactInstanceManager.onHostPause(ReactInstanceManager.java:519)
        at com.facebook.react.ReactDelegate.onHostPause(ReactDelegate.java:62)
        at com.facebook.react.ReactActivityDelegate.onPause(ReactActivityDelegate.java:93)
        at com.facebook.react.ReactActivity.onPause(ReactActivity.java:50)
        at android.app.Activity.performPause(Activity.java:6819)
        at android.app.Instrumentation.callActivityOnPause(Instrumentation.java:1322)
        at android.app.ActivityThread.performPauseActivityIfNeeded(ActivityThread.java:3740)
        at android.app.ActivityThread.performPauseActivity(ActivityThread.java:3717)
        at android.app.ActivityThread.performPauseActivity(ActivityThread.java:3691)
        at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:3665)
        at android.app.ActivityThread.-wrap16(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1489)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6119)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
2019-12-30 11:31:24.551 6821-6869/com.x.x E/AndroidRuntime: FATAL EXCEPTION: mqt_native_modules
    Process: com.x.x, PID: 6821
    java.lang.IllegalArgumentException: setDataSource failed: status = 0xFFFFFFFF
        at wseemann.media.FFmpegMediaMetadataRetriever.setDataSource(Native Method)
        at com.shahenlibrary.Trimmer.Trimmer.getPreviewImageAtPosition(Trimmer.java:550)
        at com.shahenlibrary.Trimmer.TrimmerManager.getPreviewImageAtPosition(TrimmerManager.java:84)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:371)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:150)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:26)
        at android.os.Looper.loop(Looper.java:154)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:225)
        at java.lang.Thread.run(Thread.java:761)
</pre>

cristianoccazinsp, 2019-12-30:
That's an interesting crash, what device did you use? I have never seen that issue on the test devices I have, and the recording just stops fine when the app is minimized without any manual call to stop.  Tested on both a slow and a fast device.

Have you tried removing the `captureAudio` prop? Can you reproduce the issue consistently every time or does it happen randomly?

cristianoccazinsp, 2019-12-30:
Also, does the app really crash? Looking at the code, there's a try/catch so it should be a harmless warning only:
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L214

md-cherepanov, 2019-12-30:
@cristianoccazinsp 

> Also, does the app really crash? Looking at the code, there's a try/catch so it should be a harmless warning only.

You are right. Thanks to your leading question, I examined a logs once more and figured out that it was react-native-video-processing which caused crash (because it could not take screenshot from such short video).

Thank you for your help.

closed 2019-12-30 by md-cherepanov