react-native-camera/react-native-camera #2663: Build fails on android with react-native 0.57.1

VilleMiekkoja, 2020-01-01:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 8.1.0
- react-native-camera: 3.15.1
- react-native: 0.57.1
- react: 16.6.0-alpha.8af6728
- react-navigation: 3.11.1

**Description/Current Behaviour**

Launching an application with react-native-camera@3.15.1 fails in build phase with following message:

> Task :react-native-gesture-handler:compileDebugJavaWithJavac FAILED

> ...node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerEvent.java:3: error: package android.support.v4.util does not exist. import android.support.v4.util.Pools;

> ...node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerButtonViewManager.java uses or overrides a deprecated API.

**Expected Behaviour**

Expect build to succeed, as it does when I build with react-native-camera@2.11.2.

So with version 2.11.2, it builds ok. What is weird is that this react-native-camera does not even have react-native-gesture-handler as its dependencies. Still for some reason, it fails for that if I apply the latest version of this library, and works with version 2.11.2.

**Steps to Reproduce**

Build react-native project for android with following dependencies:
    "react": "16.6.0-alpha.8af6728",
    "react-native": "^0.57.1",
    "react-native-camera": "^3.15.1",
    "react-native-gesture-handler": "1.0.17",
    "react-navigation": "^3.11.1",

No expo used here.

VilleMiekkoja, 2020-01-01:
Tried to reverse jetify this dependency with npx jetify -r. But after that, build fails with error:

node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraView.java:10: error: package android.support.v4.content does not exist
import android.support.v4.content.ContextCompat

Also errors like these then appear:

node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java:31: error: cannot find symbol
      {"int", ExifInterface.TAG_BITS_PER_SAMPLE},

Does this mean that the library v3 does not support react-native < 60 atm?

cristianoccazinsp, 2020-01-02:
Are you sure you don't have any misconfiguration? `android.support.v4.content.ContextCompat` should be available if you're using the right code. 

I would still highly recommend migrating to RN > 60. Can't help much here since I've never used the reverse jetify.

fabriziobertoglio1987, 2020-01-13:
https://github.com/facebook/react-native/issues/25173

fabriziobertoglio1987, 2020-03-13:
react native camera supports now `"react-native": "^0.59.1"`. I am closing the issue for now. Thanks

closed 2020-03-13 by fabriziobertoglio1987