react-native-camera/react-native-camera #2664: Video is not working on iPhone XR iOS version 13.1.x

nitinaloha, 2020-01-02:
# Video recording is not working on iPhone XR iOS version 13.1.x

Could you please assist us regarding this issue.

Also the codec property does not work for iOS.

cristianoccazinsp, 2020-01-02:
I'm pretty sure `options['codec'] = RNCamera.Constants.VideoCodec['H264']` is ignored/useless with iOS 13. iOS will refuse to record in anything except hevc.

What error are you getting?

nitinaloha, 2020-01-03:
Thank you for your quick repsonse. 
The video recording is working fine for iPhone XR iOS version 13.1.x now.
The only problem I am facing is with using the "codec" across different iPhone devices to reduce the video size along with the "videoBitrate" property.
iPhone X and XR always use HVEC codec and do not use H264.
Is there any work around to fix this so that we get consistent video size throughout all iPhone devices.
Just FYI the "videoBitrate" property works very great with Android.
Thank you

cristianoccazinsp, 2020-01-03:
I haven't been able to figure out the HVEC issue neither, looks like Apple won't let you use H264 at all. Let me know if you find something about it.

nitinaloha, 2020-01-03:
Thank you
I tried using "HVEC" codec which it is working for iOS X and XR to reduce the video size along with the "videoBitrate" property.
But I face issues for lower devices such as iPhone 6 which in turn increases the video size.
Need to figure out or get some help from the react native camera team.

cristianoccazinsp, 2020-01-03:
Different devices will record at different qualities and sizes. Your bet bet is to use a lower resolution (you can go as low as 288p with iOS which gives you 2-3 minutes videos in 20mb)

nitinaloha, 2020-01-03:
So again, if I use 288p quality for iOS, which codec should I use to support all iPhone devices to reduce the videoBitrate?

cristianoccazinsp, 2020-01-03:
The codec doesn't really matter since right now it will be ignored in most devices running iOS 13.x. 

The bitrate I don't really know, I haven't used it all, but have seen people having issues with it so I don't think it changes much. You still want to squeeze more storage, you might be better off adding some post-processing to the video.

nitinaloha, 2020-01-03:
Okay I will try some way. Thank you for all your help and suggestions. I really appreciate.

cristianoccazinsp, 2020-01-03:
Let me know of your findings! I have been fighting HVEC and video sizes for some time but gave up, Apple wins this time.

nitinaloha, 2020-01-10:
Hi All,
I have found a work around for time being. Here is the code 
`const options = { 
                    quality     : RNCamera.Constants.VideoQuality['4:3'],
                    videoBitrate: 0.8 * 1000 * 1000
                };
                // Get the device dimension
                const dim = Dimensions.get('window');

                if (Platform.OS === 'ios') {
                    const majorVersionIOS = parseInt(Platform.Version, 10);

                    if (majorVersionIOS >= 11) {
                        if (dim.width >= 800) {
                            options['codec'] = RNCamera.Constants.VideoCodec['HVEC'];
                        } else {
                            options['codec'] = RNCamera.Constants.VideoCodec['H264'];
                        }
                    } else {
                        options['codec'] = RNCamera.Constants.VideoCodec['H264'];
                    }
                }
                const promise = this.camera.recordAsync(options);
                const data = await promise;`

Now we check the iOS version and the iPhone device using the device dimensions. According to these condition above we apply the codec 'HVEC' for higher iPhone devices (X and above) and codec 'H264' for lower iPhone devices (below X).

This works great for me. Let me know if this works for others too.
Thank you

cristianoccazinsp, 2020-01-10:
What's the different in doing this and just forcing H264 all the time? I believe 'H264' is ignored if it's not supported anyways.

nitinaloha, 2020-01-13:
This works for me as expected.
H264 works for lower devices and HVEC works for higher devices as mentioned in previous comment.
You can try using and see the results for yourself.

mcorner, 2020-01-29:
Is it really not possible to force H264 all of the time?  I feel like I had this working in a previous version of our app (even on ios13) and I am running into trouble forcing it to happen now.  This is problematic because we record video and upload it and display it in a browser (and Chrome etc can't play hvec).  So we will have to transcode it which is just a pain.

cristianoccazinsp, 2020-01-29:
If you ever find a way to do it, please let me know! I've spent some time
with it but no luck, looks like iOS 13 just forces you to use it.

Perhaps the transcoding could be done on the device?

mcorner, 2020-01-29:
We set the codec (H264) and quality (720p) in options and the defaultVideoQuality (720p) in the component consistent with #1858   This is working in our production build on an XR with ios13 (it produces H264), but not in our latest build (upgraded many things).  So ATM I can't explain it.

We could probably transcode locally, but that will probably be slow.  We will keep looking at it.

nitinaloha, 2020-01-29:
Below code works for 
                **const options = { 
                    quality     : RNCamera.Constants.VideoQuality["4:3"],
                    videoBitrate: 0.8 * 1000 * 1000
                };
                // Get the device dimension
                const dim = Dimensions.get('window');

                if (Platform.OS === 'ios') {
                    const majorVersionIOS = parseInt(Platform.Version, 10);

                    if (majorVersionIOS >= 11) {
                        if (dim.width >= 800) {
                            options['codec'] = RNCamera.Constants.VideoCodec['HVEC'];
                        } else {
                            options['codec'] = RNCamera.Constants.VideoCodec['H264'];
                        }
                    } else {
                        options['codec'] = RNCamera.Constants.VideoCodec['H264'];
                    }
                }
                // Get camera cache data
                const promise = this.camera.recordAsync(options);
                const data = await promise;

                <RNCamera
                        ref={ref => {
                            this.camera = ref;
                        }}
                        style={styles.preview}
                        type={RNCamera.Constants.Type.back}
                        androidCameraPermissionOptions={null}
                        androidRecordAudioPermissionOptions={null}
                        defaultVideoQuality={RNCamera.Constants.VideoQuality["4:3"]}
                 >
**

If anyone wants they can try this and test.

mcorner, 2020-01-29:
That code is going to encode using HVEC sometimes.  The point is that some people would like to *force* it to *always* use H264.  (For instance if you have an app that uploads a video to be viewed in Chrome, you can't use HVEC without transcoding it).

mcorner, 2020-01-29:
We did some binary search and discovered that 3.6.0 seems to work and 3.7.0 does not.  There appears to be a race condition that was introduced in 3.7.0 where the availableCodecs come back as [hvc1] (https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1047).  But when we introduced some logging into that part of the code it returns [hvc1,avc1,jpeg].  (iphone XR on ios13)

So that smells like some kind of race condition in setting the quality.  I am sorry I don't have a better lead than that, but we will use 3.6.0 until someone can diagnose it better.

cristianoccazinsp, 2020-01-30:
Why do you think it is a race condition? Doesn't that code run only when `recordAsync` is called, and is not called concurrently at all? Maybe moving that whole code block further down the method (inside the dispatch async call) would resolve it? I still don't see where's the race condition though, `availableVideoCodecTypes ` should report what's available.  I will help with the testing once I get some time tomorrow since this issue affects us as well and I thought it was just a limitation with iOS 13.

mcorner, 2020-01-30:
@cristianoccazinsp Because we get two different results for availableCodecs without changing anything but inserting a few log statements.  I don't know *where* the race is, but there are a lot of asynchronous blocks in the code.  availableCodecs is based on your selected quality, but we have exactly the same selected (and default) quality (720p).

mcorner, 2020-01-30:
Something that "smells bad" to me is that when it sets the quality asynchronously wrt checking the availableCodecs.

I don't see anything that would prevent the following sequence:

A) https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1024

then: 
 
B) https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1047

and then:

C) https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1501

A different pattern could have been made more likely when we inserted debug statements that slows down the execution of B, creating the correct sequence A, C, B.  But...I don't know this code except what I have just recently read, so just an idea.

cristianoccazinsp, 2020-01-30:
I cannot seem to reproduce this on an iphone 7 with iOS 13 (my only currently available test device). But something easy to try if this is really a race condition, would be to move the following block, into the dispatch async block.

Move https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1044  into https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1101

That should definitely get rid of any race condition. It might make sense that changing the session preset also affects the `movieFileOutput.availableVideoCodecTypes` value. 

Lastly, the "only available in ios 10" warning log seems to also be off (wrong if). 

I'm willing to give that change a try but I can't confirm it right now, unless anyone would be up to test it :)

cristianoccazinsp, 2020-01-30:
@mcorner can you test with this? https://github.com/cristianoccazinsp/react-native-camera/tree/ios-codec-fix

Update basically moves codecs, max duration, and max file size to the session queue so it should prevent any race condition due to session/preset updates. If the update works, I will send the PR.

mcorner, 2020-01-30:
@cristianoccazinsp Oh, this is awesome.  This appears to fix the issue for us.  You might want someone else to test it (someone with an XR XS etc), but we will continue using your branch until you merge.  I will let you know if it gives us any problems.  Thanks!!

cristianoccazinsp, 2020-01-30:
So you were right about the race condition. It might take us some more time to test it thoroughly since we have to go through all the testflight garbage but we will also be testing it soon :)

cristianoccazinsp, 2020-01-30:
PR here: https://github.com/react-native-community/react-native-camera/pull/2694
Might help with more review. I will be testing this today and tomorrow on a few devices through testflight.

cristianoccazinsp, 2020-02-03:
Tested on an iPhone 11 pro 11 and the PR definitely fixes it! Videos are now uploaded with the right (browser playable) codec. Thanks for finding this out @mcorner

nitinaloha, 2020-02-03:
Please merge this PR we are waiting for this fix. This is going to resolve more than one issue for different developers.
Thanks alot

closed 2020-02-05 by sibelius

n1ru4l, 2020-02-05:
:tada: This issue has been resolved in version 3.17.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.17.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

RandalNyentek, 2020-05-01:
I have an iPhone X running 13.4.1 and it is not honoring the h264 codec option. I've tried going into my camera settings and tested Most Compatible and High Efficiency camera formats. My videos are still HEVC format.

Here are the options I'm passing to recordAsync

`
this.videoRecordOptions = {
quality: RNCamera.Constants.VideoQuality['480p'],
orientation: RNCamera.Constants.Orientation.portrait,
maxDuration: 60,
path: this.videoFilePath,
codec: RNCamera.Constants.VideoCodec['H264']
}
`

cristianoccazinsp, 2020-05-01:
When I tested this, it was definitely working. Unless something changed with iOS 13.4.1 or some RNCamera update. I don't have a device to test right now.

RandalNyentek, 2020-05-01:
Thanks for your reply. I'll keep looking into my issue, and if anything comes up I'll post back.

mcorner, 2020-05-01:
Are you also setting the default quality in the props for the component?
 <RNCamera defaultVideoQuality=RNCamera.Constants.VideoQuality['480p'] ...

RandalNyentek, 2020-05-01:
Yes. Here is my RNCamera Component from my render method.

<RNCamera
					ref={ref => {
						this.camera = ref
					}}
					defaultVideoQuality={RNCamera.Constants.VideoQuality['480p']}
					captureAudio={true}
					keepAudioSession={true}
					style={{ flex: 1 }}
					type={this.state.cameraType}
					flashMode={RNCamera.Constants.FlashMode.auto}
					permissionDialogTitle={'?'}
					permissionDialogMessage={'?'}
					orientation={RNCamera.Constants.Orientation.portrait}
				/>