react-native-camera/react-native-camera #2665: takePicture failed

lucasferreiraestevam, 2020-01-06:
Android : 9
react-native-camera : v3.15.1
react-native : 0.61.5

Possible Unhandled Promise Rejection (id: 1) : Error: takePicture failed...

https://codepen.io/lucasferreiraestevam/pen/MWYrQBj.js

[image]

lucasferreiraestevam, 2020-01-07:
Solved !

____________

Only on android if setState is called before the takePictureAsync function sometimes RNcamera returns Possible Unhandled Promise Rejection (id: 1): Error: takePicture failed ...

pereirafabiosantos2018, 2020-10-16:
This is happening to me on Android device.

pereirafabiosantos2018, 2020-10-16:
When I try press the button to open camera, the following error occurs:
error Error: takePicture failed
    at Object.fn [as takePicture]

closed 2021-01-06 by stale[bot]

NguyenHuuManh, 2021-11-05:
i have same problem, please help me!