react-native-camera/react-native-camera #2666: I'm curious about this file.

savetheearth2, 2020-01-07:
# Question

**To Do First**
- [O] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [O] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [O] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

Hello! I am not good at English.

My question is about the react-native-camera file.
I followed the Installation section of the following link.
[](https://react-native-community.github.io/react-native-camera/docs/installation.html)

I've read several related documents, but I'm not sure what they mean.
I would like to try the PinchGestureHandler function included in ZoomView.js.
Can the examples included in "react-native-camera" be executed as an expo client?
I wonder how to implement it.
Should I run as an Android studio?

Thank you

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

fabriziobertoglio1987, 2020-01-13:
You should try to implement the functionality as [expo includes already `react-native-gesture-handler`](https://docs.expo.io/versions/latest/sdk/gesture-handler/), which is the library that provides the `PinchGestureHandler` and `State` Components.

For more information about [`PinchGestureHandler`](https://software-mansion.github.io/react-native-gesture-handler/docs/handler-pinch.html)

as you can see in the [expo `Camera` docs](https://docs.expo.io/versions/latest/sdk/camera/) the zoom props that you use to build this functionality is available

savetheearth2, 2020-01-15:
> You should try to implement the functionality as [expo includes already `react-native-gesture-handler`](https://docs.expo.io/versions/latest/sdk/gesture-handler/), which is the library that provides [...]

Thank you very much for your reply!
I'm curious about how to install and execute the "react-native-camera" file of "react-native-community."
https://react-native-community.github.io/react-native-camera/docs/installation.html
Which folder should I install this link in?

fabriziobertoglio1987, 2020-01-15:
follow the instructions on the [expo docs](https://docs.expo.io/versions/latest/sdk/camera/). If you do not understand them search for an expo training

savetheearth2, 2020-01-16:
> follow the instructions on the [expo docs](https://docs.expo.io/versions/latest/sdk/camera/). If you do not understand them search for an expo training

Thank you very much for your reply!
I first wonder if you are talking about the project on this link.
I want to run this project that includes ZoomView.js.
https://github.com/react-native-community/react-native-camera.git

fabriziobertoglio1987, 2020-01-16:
you can test the project from the examples/advanced folder
clone the repo and `cd examples/advanced/advanced && yarn`

the app is using the ZoomView, but it is laggy and not worth it.
Just add that code in your Expo project, **read the docs**

search the issues on github **pinch to zoom**

savetheearth2, 2020-01-18:
> you can test the project from the examples/advanced folder
> clone the repo and `cd examples/advanced/advanced && yarn`
> 
> the app is using the ZoomView, but it is not native animation.
> Just add that code in your Expo project, **read the docs**
> 
> search the issues on github **pinch to zoom**

Thank you!
I'll have to try

closed 2020-03-25 by stale[bot]