react-native-camera/react-native-camera #2667: Black screen on second instanciation

AzRunRCE, 2020-01-07:
# Bug Report

**Platforms**
Looked on Android 5 and 10

**Versions**
Android only (simulator and device)

- Android: 10 & 5 sure
- react-native-camera: ^3.15.1
- react-native: 0.61.5
- react:16.9.0
- react-navigation: ^4.0.10
- react-navigation-material-bottom-tabs: ^2.1.5
- react-navigation-tabs:2.7.0

**Description/Current Behaviour**
Hello guys and thank you for your work.
When we use a bottomTabsView with **createBottomTabNavigator** to create 2 tabs. At first instanciation the android authorisation is asked and it's work. We can see the video capture. But if go into another tabs and i come back into the initial tabs where the RNCamera is viewed **it's give me a black screen at the second use.**
sorry for my bad english.

**Expected Behaviour**

Stream/Show the video capture

**Steps to Reproduce**
1) react-native init poc-react-native-camera-issue
2) install dependancies 
    - react-native-camera
    - react-native-gesture-handler
    - react-native-paper
    - react-native-reanimated
    - react-native-vector-icons
    - react-navigation
    - react-navigation-material-bottom-tabs
    - react-navigation-tabs
3) Add android permission
4) Paste this content -> _https://pastebin.com/raw/Ur2k8VMG_

1) Go Profile tab
2) Acept authorisation if asked
3) Come back into Home
4) Go to Profil tab
5) Result black / no stream camera capture

sibelius, 2020-01-07:
you need to stop one camera to show another

antoninleissen, 2020-01-08:
@sibelius How can we do that ? Didn't find on doc.

Dlessi, 2020-01-10:
Same issue Android 9

khat33b, 2020-01-13:
@sibelius Could you please explain how to do this? I tried removing the camera when the page is not in focus as shown [here](https://github.com/react-native-community/react-native-camera/blob/master/docs/react-navigation.md) but the problem persists.

moniquealtero, 2020-01-15:
> 
> 
> you need to stop one camera to show another

I'd like to know how to do this. 

I'm treating the 'new' instance with NavigationEvents onDidFocus, but i'm not able to 'refresh' the camera.

moniquealtero, 2020-01-15:
After a quick research i found this:

**pausePreview:** void
_Pauses the preview. The preview can be resumed again by using resumePreview().

**resumePreview**: void_
_Resumes the preview after pausePreview() has been called._

https://react-native-community.github.io/react-native-camera/docs/rncamera#pausepreview-void

yauheniivashkevich, 2020-01-28:
I think you can use Facc as a solution
https://react-native-community.github.io/react-native-camera/docs/rncamera#facc-function-as-child-components

closed 2020-02-01 by AzRunRCE