react-native-camera/react-native-camera #2670: Barcode/Text feature bounds incorrect on rotation without change in orientation

TimFinucane, 2020-01-11:
# Bug Report

- Did you try latest release? Yes - 3.15.1
- Did you try master? No
- Did you look for existing matching issues? Yes

**Platforms**

Tested on Android Only

- Android: 10, Samsung S10+, One UI 2.0
- react-native-camera: 13.5.1
- react-native: 0.61.5
- react: 16.9.0
- react-navigation: 4.0.10

**Description/Current Behaviour**

On rotating a phone with One UI, without auto-rotate enabled, the user can press a button to change the phones orientation otherwise nothing happens.

When rotating the phone and NOT pressing the button, the bounds for barcodes and text scanned by RNCamera (haven't tested facial features) are rotated anyway, causing them to be in incorrect positions on the screen.

**Expected Behaviour**

When the camera is rotated but orientation has not changed, feature positions do not get rotated.

**Steps to Reproduce**

- Enable google barcode detection, and record the position
- Check how the position changes normally (pitching up and down changes y dimension, etc)
- Rotate the device WITHOUT changing orientation, then check position again, this time
pitching up and down changes the x dimension.

**Additionals**

My explanation may be a bit confusing sorry. I have attached a video that shows more clearly what is happening.

In the video I first move the camera whilst the device is vertical. Then I rotate the device without changing orientation, and move the camera. Then I change the orientation. The yellow dot is calculated to be at the centre of the bounds given by onGoogleVisionBarcodesDetected.

[Link to video](https://thumbs.gfycat.com/UnawareAcrobaticBedbug-mobile.mp4)

I've tested this with onGoogleVisionBarcodesDetected, and onTextRecognized

whtswrng, 2020-01-28:
Can somebody look into this?

sibelius, 2020-06-05:
cc @renatobenks

xiic, 2020-10-15:
Having a similar issue (only with MLKit/Google Vision) - even if keeping the same device orientation, one of the coordinates is substantially off. By how much strangely seem to depend on the preview size. Is it possible to somehow get the android preview size? Is somebody looking into this?

closed 2021-01-13 by stale[bot]

chiubaka, 2021-07-22:
This is still an issue for me. Looks like it never got any attention?

renatobenks, 2021-08-01:
Hey @chiubaka, sorry for anyone hasn't paid attention to this for so long. However, I tried to reproduce it and find any workaround a long time ago but unfortunately, I couldn't.

I can manage it one more shot to try to fix it, but it seems to be happening in a third-party module from Google which we used to detect and scan barcodes and texts. Maybe, I miss something between the native modules, in somewhere in our side, so I can try a fresh start on that, but I can't promise anything, sorry!

chiubaka, 2021-11-19:
I did some poking and and I'm pretty sure the problem isn't in a third-party module.

The issue is that the [`getCorrectCameraRotation` method in `RNCameraViewHelper.java`](https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L339-L347) doesn't account for the current screen orientation, so even if the screen's orientation is locked we're [passing a rotation value down to the detector tasks](https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L164) as if the screen's orientation _weren't_ locked.

If I hardcode the rotation value, I can get this to behave properly as if the screen were locked to a specific orientation, which further proves to me that the issue isn't in third-party modules.

The difficulty is that "natural orientation" varies from device to device. E.g. My Pixel 3 registers a portrait orientation as rotation=0, since it's natural for the device to be in portrait mode, while a tablet might register landscape orientation as rotation=0. I have found [a few answers on Stack Overflow](https://stackoverflow.com/a/9888357/599391) that hint at ways to determine if a device's natural orientation is portrait or landscape, but I'm unsure if there are some devices where the natural landscape orientation is the reverse of what it is on other devices (i.e. 90 degrees rotated from portrait vs 270 degrees rotated from portrait).

The other difficulty is that there isn't a built-in way on Android to tell if the current screen orientation is landscape left vs landscape right or portrait vs portrait upside-down, but if you could get reliable "natural orientation", you can determine which orientation the device is actually in by combining the device rotation value with the screen orientation reported by the configuration.

Assuming that "natural orientation" is always either portrait or landscape left orientation, I have a PR that seems to reliably fix this issue in #3384.

@renatobenks Any chance you could give that a review?