react-native-camera/react-native-camera #2671: App crash when user navigates back from camera component on Android device JB 4.1

arifinoid, 2020-01-13:
# Bug Report

App crash when user navigates back from camera component on Android device JB 4.1

**To Do First**

- [ ] Did you try latest release? Yes I did, react-native-camera@3.15.1
- [ ] Did you try master? Yes I did.
- [ ] Did you look for existing matching issues? Yes I did 

**Platforms**

Android emulator with API 16 Jelly bean 4.1

**Versions**
Jelly bean 4.1.

- Android: emulator Jelly bean 4.1
- react-native-camera: react-native-camera@3.15.1
- react-native: "react-native": "^0.59.2",
- react: "react": "16.8.3",
- react-navigation: "react-navigation": "3.7.1",

Here is the log from adb logcat
```
D/dalvikvm( 5366): GC_FOR_ALLOC freed 1821K, 7% free 41095K/44103K, paused 17ms, total 17ms
V/EmulatedCamera_CallbackNotifier( 1173): enableMessage: msg_type = 0x10
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_FRAME
V/EmulatedCamera_CallbackNotifier( 1173): **** Currently enabled messages:
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ERROR
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_FOCUS
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ZOOM
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_FRAME
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_METADATA
V/EmulatedCamera_CallbackNotifier( 1173): disableMessage: msg_type = 0x10
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_FRAME
V/EmulatedCamera_CallbackNotifier( 1173): **** Currently enabled messages:
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ERROR
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_FOCUS
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ZOOM
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_METADATA
V/EmulatedCamera_Camera( 1173): doStopPreview
V/EmulatedCamera_Device( 1173): stopDeliveringFrames
V/EmulatedCamera_Device( 1173): stopWorkerThread
V/EmulatedCamera_Device( 1173): Stopping emulated camera device's worker thread...
V/EmulatedCamera_Device( 1173): Emulated camera device's worker thread has been stopped.
V/EmulatedCamera_QemuDevice( 1173): stopDevice
V/EmulatedCamera_QemuDevice( 1173): stopDevice: Qemu camera device '0x8020000005ac8514' is stopped
V/EmulatedCamera_Preview( 1173): stopPreview
V/EmulatedCamera_CallbackNotifier( 1173): disableMessage: msg_type = 0x10
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_FRAME
V/EmulatedCamera_CallbackNotifier( 1173): **** Currently enabled messages:
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ERROR
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_FOCUS
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ZOOM
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_METADATA
V/EmulatedCamera_CallbackNotifier( 1173): disableMessage: msg_type = 0x10
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_FRAME
V/EmulatedCamera_CallbackNotifier( 1173): **** Currently enabled messages:
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ERROR
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_FOCUS
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ZOOM
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_METADATA
V/EmulatedCamera_CallbackNotifier( 1173): disableMessage: msg_type = 0xffff
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ERROR
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_SHUTTER
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_FOCUS
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_ZOOM
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_PREVIEW_FRAME
V/EmulatedCamera_CallbackNotifier( 1173):     CAMERA_MSG_VIDEO_FRAME
[... 42 lines cut ...]
E/AndroidRuntime( 5366): 	at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
E/AndroidRuntime( 5366): 	at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:166)
E/AndroidRuntime( 5366): 	at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
E/AndroidRuntime( 5366): 	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:723)
E/AndroidRuntime( 5366): 	at android.view.Choreographer.doCallbacks(Choreographer.java:555)
E/AndroidRuntime( 5366): 	at android.view.Choreographer.doFrame(Choreographer.java:524)
E/AndroidRuntime( 5366): 	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:711)
E/AndroidRuntime( 5366): 	at android.os.Handler.handleCallback(Handler.java:615)
E/AndroidRuntime( 5366): 	at android.os.Handler.dispatchMessage(Handler.java:92)
E/AndroidRuntime( 5366): 	at android.os.Looper.loop(Looper.java:137)
E/AndroidRuntime( 5366): 	at android.app.ActivityThread.main(ActivityThread.java:4745)
E/AndroidRuntime( 5366): 	at java.lang.reflect.Method.invokeNative(Native Method)
E/AndroidRuntime( 5366): 	at java.lang.reflect.Method.invoke(Method.java:511)
E/AndroidRuntime( 5366): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
E/AndroidRuntime( 5366): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
E/AndroidRuntime( 5366): 	at dalvik.system.NativeStart.main(Native Method)
W/ActivityManager( 1421):   Force finishing activity com.sicepat.pod/.MainActivity
W/AudioFlinger( 1173): session id 51 not found for pid 1173
D/dalvikvm( 5366): GC_CONCURRENT freed 2581K, 9% free 40447K/44103K, paused 1ms+0ms, total 11ms
D/gralloc_ranchu( 1170): gralloc_alloc: Creating ashmem region of size 1458176
W/ActivityManager( 1421): Activity pause timeout for ActivityRecord{9fcb8a10 com.sicepat.pod/.MainActivity}
D/gralloc_ranchu( 1170): gralloc_alloc: Creating ashmem region of size 1458176
D/gralloc_ranchu( 1170): gralloc_alloc: Creating ashmem region of size 1458176
I/Process ( 5366): Sending signal. PID: 5366 SIG: 9
```

cristianoccazinsp, 2020-01-13:
`E/AndroidRuntime( 5366): java.lang.NoSuchMethodError: android.os.HandlerThread.quitSafely`

Traces back to: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/CameraView.java#L149

https://developer.android.com/reference/android/os/HandlerThread#quitSafely() (added in Android 4.3 (API level 18))

So the issue is that Android 4.1 does not have the above method.

Possible fix: Conditionally call `quit` if SDK version < 18. Considering how old Android 4.1 is, I would suggest raising the minSdk version to 18 in the meantime.

cristianoccazinsp, 2020-01-13:
@arifinoid  can you test https://github.com/react-native-community/react-native-camera/pull/2674 ? Should fix the crash.

arifinoid, 2020-01-14:
@cristianoccazinsp Okay, I'll test it .. thank you

arifinoid, 2020-01-14:
@cristianocca @cristianoccazinsp Thank you so much, it works as well..

closed 2020-01-14 by arifinoid