react-native-camera/react-native-camera #2672: Could not find method compile() for arguments [com.android.support:recyclerview-v7:25.3.1]

palashmandokhot, 2020-01-13:
hi,

In my android studio build gradle is failed i dont know how so then i check the log and i saw this error Error:Could not find method compile() for arguments [com.android.support:recyclerview-v7:25.3.1] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

Please install the Android Support Repository from the Android SDK Manager. Open Android SDK Manager

then i check it but android support respository is already installed version 47.0.0 i restart android studio but it shows same error how i solve this error from android studio please help me to solve this problem?

fabriziobertoglio1987, 2020-01-13:
here is a good explanation of [the difference between `compile` and `implement`](https://stackoverflow.com/a/48388463/7295772). Please show us the code where you try to compile `com.android.support:recyclerview-v7:25.3.1`. 

You will find it in the `build.gradle` files or make a search for those keywords

additionally you can read [Could not find method compile() for arguments Gradle](https://stackoverflow.com/questions/23796404/could-not-find-method-compile-for-arguments-gradle)

palashmandokhot, 2020-01-14:
hi,
thank you for you replying,yeah i send it but i am little hesitate that its not good or great for me if i shared my project code snippets because if anyone misuse it because its a public so can i share my code snippets in private?

palashmandokhot, 2020-01-17:
hi,
check the link everything is setup.

closed 2020-03-24 by stale[bot]