react-native-camera/react-native-camera #2673: rashes on iphone X

alex-mironov, 2020-01-13:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2020-01-13:
Any additional info? Hard to debug with just that. Do you see this very often? Can you trace the crash with XCode and get the actual line of the crash? Might take up to 24 hours for the crash logs to be sent from Apple to XCode.

alex-mironov, 2020-01-17:
Sorry for the long reply.
In my case, the issue was self-explanatory, when I debugged the app through Xcode.
```
[warn][tid:com.facebook.react.ShadowQueue][RNCameraManager.m:465] Checking audio permissions without having key 'NSMicrophoneUsageDescription' defined in your Info.plist. Audio Recording for your video files is therefore disabled. If you do not need audio on your recordings is is recommended to set the 'captureAudio' property on your component instance to 'false', otherwise you will have to add the key 'NSMicrophoneUsageDescription' to your Info.plist. If you do not your app will crash when being built in release mode. You can learn more about adding permissions here: https://stackoverflow.com/a/38498347/4202031
2020-01-17 22:26:29.721062+0200 farutexMobile[7059:289006] [access] This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data.

```
What confused me, it worked fine on my iPhone 6 but was crashing on newer ones.
Fix is here https://stackoverflow.com/a/40347633/1014186

closed 2020-01-17 by alex-mironov