react-native-camera/react-native-camera #2677: Same Codec but different file output (mp4 vs mov)

enigmablue, 2020-01-17:
Hello, i have the below code for my options for takepicture async

```
  let data = await this.camera.recordAsync({
        quality: RNCamera.Constants.VideoQuality['480p'],
        // videoBitrate: 300*1000,
        videoBitrate: 600*1000,
        // quality: RNCamera.Constants.VideoQuality['4:3'],

        mirrorVideo: true
      }).then((s)=>{
        this.camera.pausePreview()
        data = {...s,type : "video"}
        console.log("data to send", data)
```

I took two videos. The first one was a video with the back camera (taking a picture of a scene). Whilst the second one was with the front camera (selfie) with me speaking. However the 2nd one came out with a .mov. They both have the same codec. 

Why is the file format different? And how to make it consistent?  Thanks!

data to send {"codec": "avc1", "deviceOrientation": 1, "isRecordingInterrupted": false, "type": "video", "uri": "file:///var/mobile/Containers/Data/Application/0507A3C9-7F8C-4C3E-BE02-521F3E5E748C/Library/CachesCameraFlip/8051EFDC-BF3D-44C3-9897-DE1BB48B4E2D.**mp4**", "videoOrientation": 1}

data to send {"codec": "avc1", "deviceOrientation": 1, "isRecordingInterrupted": false, "type": "video", "uri": "file:///var/mobile/Containers/Data/Application/0507A3C9-7F8C-4C3E-BE02-521F3E5E748C/Library/Caches/Camera/28F3878D-8D32-4FF2-85BC-0CD3D7938629.**mov**", "videoOrientation": 1}

cristianoccazinsp, 2020-01-17:
Interesting. Aren't all iphone videos .mov? Have never experienced this since I manually set the file name on the videos I record in order to be able to decide its location.

Looking at the folder location, looks like it might be related to the video mirroring (look at the folder `CachesCameraFlip` vs `Caches/Camera`).

Further digging, here's the culprit: https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1874

The above line sets a different folder, file extension, and (if I'm not wrong), even **hardcodes** a video quality. So by just mirroring the video you will get a much lower quality recording. I don't really know the rationale behind that behaviour though.

enigmablue, 2020-01-18:
thanks! i will check the flag out.

I believe the mirror was set because android had some funny issues. I will double check too.

enigmablue, 2020-01-18:
After i removed the mirror it looks good. I dont think i needed it in first place. Weird. 
Thanks @cristianocca

cristianoccazinsp, 2020-01-18:
I believe the above is still a bug if someone wants to fix it. Looks like mirroring video is a bit broken right now. Might want to leave the issue open in case someone wants to tackle it!

nica0012, 2020-05-05:
Still broken

closed 2020-07-11 by stale[bot]

mattgle, 2021-02-05:
Still broken dude!!!!!! I need help!