react-native-camera/react-native-camera #2678: ios GoogleVisionBarcodeDetection.BarcodeMode.AlTERNATE doesn´t work

grean, 2020-01-19:
When i use this parameter with RNCamera/MLkit on ios like this : 
`googleVisionBarcodeMode={RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeMode.ALTERNATE}`

i've got this error : 
`typeError: Cannot read property 'ALTERNATE'of undefined`

In Android it works

This error happens on a real device iPad 7gen
With RN 0.61.5
and React-native-camera 3.15.1

But, if a remove the prop, everything works well even if i scan vertically or horizontally.

manishkumar03, 2020-06-02:
I am facing the same issue in iOS app. Are there any plans to add the support for inverted barcodes in iOS?

interstates21, 2020-06-11:
Same issue

manishkumar03, 2020-06-11:
The support for this has been added to iOS now. If it still doesn't work, please let me know.

renatobenks, 2020-06-12:
which version are you using @interstates21? try to update to the latest one, because as @manishkumar03 told you above, he introduced that support in https://github.com/react-native-community/react-native-camera/pull/2851, so if you use the latest, or since 3.27.1 version, it should work

closed 2020-06-12 by renatobenks