react-native-camera/react-native-camera #2679: Do I have to implement my own code for camera permission access on iOS?

jackblackCH, 2020-01-19:
1. Do I have to implement my own code for camera permission access on iOS?
It's confusing as none of the examples implemented a camera permission access for iOS, nor could I find anything in the docs.

2. Does the library automatically show the ios camera permission dialog?

MateusAndrade, 2020-01-21:
As i remember, on iOS you should specify at your Info.plist you request to camera. Also, by requesting access to camera this should call the permission dialog.

jackblackCH, 2020-01-21:
Thanks. I also solved my issue in the meantime. It's confusing for newbies, thinking that you can implement any react native module into a managed expo project.

So for others:
If you set up your project with Expo, all react-native modules with real native features, i.e camera access won't work out-of-the-box. That's why there are premade expo packages made for that. I.e instead of `react-native-camera` you must use `expo-camera` and to work with the file-system or media, you'd use the MediaLibrary from expo. (`import * as MediaLibrary from 'expo-media-library'`)

closed 2020-01-21 by MateusAndrade