react-native-camera/react-native-camera #2683: [iOS] TextDetector does not recognize small and thin texts

Xrequillart, 2020-01-23:
# Bug Report

**To Do First**
- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**
Only on iOS

**Versions**
- iOS: **13.3**
- react-native-camera: **3.15.1**
- react-native: **0.59.0**
- react: **16.8.3**
- react-navigation: **3.11.1**

**Description/Current Behaviour**

The problem is only on iOS
When the text is too small or thin, the text recognition no longer detects anything. 
(rf screenshot below)
We have to zoom for it to work.

When i test on android it works very well !

**Steps to Reproduce**

 - Open ViewCamera and scan a sheet with multiple font sizes
 - Look what you return this function (`onTextRecognized`)

**Additionals**
[image]
[image]

Xrequillart, 2020-01-24:
Ok it works ! If you change the scaling parameters (file: **RNCameraUtils.m**)

```js
float scale = scale = bufferHeight>bufferWidth ? 720 / bufferWidth : 720 / bufferHeight;;
    if (position == 1) {
        scale = bufferHeight>bufferWidth ? 400 / bufferWidth : 400 / bufferHeight;
    }
```
 --- >
```js
float scale = scale = bufferHeight>bufferWidth ? 1920 / bufferWidth : 1920 / bufferHeight;;
    if (position == 1) {
        scale = bufferHeight>bufferWidth ? 1080 / bufferWidth : 1080 / bufferHeight;
    }
```
But I think it's not the most opti. What do you think ?

arousselotvi, 2020-01-29:
Is there a specific reason for scaling down the image used by MlKit that much? Because I think a lot of issues (#1706 for example) with text detection/barcode scanning/face detection on iOS might come from this.

closed 2020-04-05 by stale[bot]