react-native-camera/react-native-camera #2686: [iOS] RNCamera preview doesn't resume when app comes to foreground from background

gbalduzzi, 2020-01-27:
# Bug Report

**Platforms**
iOS

**Versions**
I noticed the issue on `3.7.0`. Upgraded to  `3.15.0` but the bug persists

- iOS:
- react-native-camera: `3.15.0`
- react-native: `0.61.2`

**Description/Current Behaviour**
With the camera working on a screen, you let the app in background and open other apps or lock the screen. After some time, reopening the app results in a black screen instead of the camera. Closing and reopening the screen does not work, the only way to make it work again is by closing and opening the app again.

I even tried to delete the camera when the app goes in background and restoring it when the app comes to foreground, without success:

```
{!this.state.inBackground && (
 <RNCamera ... />
)}

 ....

AppState.addEventListener('change', this.handleAppStateChange)

...

handleAppStateChange = (nextAppState: string) => {
    switch (nextAppState) {
      case 'active':
        this.setState({
          inBackground: false,
        })
        break
      case 'background':
      case 'inactive':
        this.setState({
          inBackground: true,
        })
        break
      default:
    }
  }
```

I found the exact same issue, but it has been closed as stale: https://github.com/react-native-community/react-native-camera/issues/2222

Note that the behaviour doesn't seem to be deterministic. Sometimes the camera restore successfully, sometimes not.

**Additionals**

I managed to reproduce the error while logging the 3 events exposed by RNCamera, without useful informations:

- `onMountError`: not called
- `onCameraReady`: when the error occurs, this is called periodically, like every 1-2 seconds
- `onStatusChange`: called once, with `cameraStatus` as `READY`

yauheniivashkevich, 2020-01-28:
please try https://react-native-community.github.io/react-native-camera/docs/rncamera#facc-function-as-child-components

cristianoccazinsp, 2020-02-03:
Interesting, I recall this being an issue but it was technically fixed, unless you are doing something else with the camera component that might not have been considered on the fix.

As opposed to the above, FACC components might not be good neither; I've had crashes of all kinds when using that, mostly related to permissions.

If you provide a reproducible demo I might try it out, but I haven't had issues since the fix. Have you tried with the latest version?

cristianoccazinsp, 2020-04-03:
I haven't observed this issue. If I had to guess, it is related to the fact that you mount/unmount on background/foreground events. That's probably conflicting with what the camera does natively.

shankar258, 2020-04-13:
I am not using the mount/unmount to set resume or pause the camera, even-though  I am getting the same error like when I sent app background and come back to the app with multiple times, camera showing black screen.

shankar258, 2020-04-13:
To reproduce the issue please follow the below steps.
1. Open camera.
2. Send the app to the background (Hit the home button)
3. Launch the app again
4. Go to step 2.
note: issue replicating in ios device

cristianoccazinsp, 2020-04-13:
The original post definitely did mount/unmount:
[image]

What version or RNCamera are you using? What device and iOS version?

shankar258, 2020-04-13:
I am using "react-native-camera": "^3.19.0" and in all iOS devices

shankar258, 2020-04-14:
Hi @cristianoccazinsp, I used the above code as mentioned by you, even-though when it went to background and came to the foreground camera is not working, please help me in this.

shankar258, 2020-04-14:
Here we are going to the background by opened the camera.

HauLuu, 2020-04-22:
Same issue here:
react-native-camera version: 3.23.1
react-native version: 0.60.0
ios version: 12.4.6

alexouille, 2020-05-20:
same issue here:
react-native-camera@3.26.0
react-native@0.62.2
iOS version: 13.4.1

cristianoccazinsp, 2020-05-20:
@alexouille  does the same happen if you run the example apps? Does it happen all the time? What device are you testing on?

alexouille, 2020-05-20:
@cristianoccazinsp not sure if I'm doing something wrong but can't run basic example, got error "Unknown argument type '__attribute__' in method -[RCTAppState getCurrentAppState:error]. Extend RCT Convert to support this type."

the black screen happens randomly after app comes back from background. Tested on iPhone 7 Plus, iPhone 8 Plus, iPhone XS Max.

cristianoccazinsp, 2020-05-20:
Odd, I haven't run the basic example in a long time. What about the advanced one?

I haven't really seen this issue anymore after it was fixed some time ago.  Are you also using react-navigation or other navigation library?

alexouille, 2020-05-20:
@cristianoccazinsp advanced example works fine, can't reproduce the problem.
yes I'm using also react-native-navigation.
I'm gonna reset my app from scratch, and add navigation & other stuff step by step to narrow things down, I'll keep you in touch when I know more.

cristianoccazinsp, 2020-05-20:
Great to see it worked.  If I had to guess, either the navigation library or your very own code might be hiding/opening the camera (or parent View) on app suspension. If that's true, it can conflict with the native implementation of the camera which already does that automatically.

alexouille, 2020-05-20:
found what was causing trouble: it was possible that in my code getUserMedia() was called twice, which ended in erratic video behaviour, like black screen, image freeze.
couldn't reproduce the problem once I ensured getUserMedia() was called only once for all.
thank you for your help.

closed 2020-07-26 by stale[bot]