react-native-camera/react-native-camera #2687: [Android] Xiaomi Mi A2 Lite recordAsync crash

livnunes, 2020-01-27:
# Bug Report

**To Do First**

- [Y] Did you try latest release?
- [N] Did you try master?
- [Y] Did you look for existing matching issues?

**Platforms**

Tested on Xiaomi Mi A2 Lite, camera gets glitched, and freezes after a few seconds, example screenshot at the bottom. Also tested with a clean 0.59.10 react native project.

**Versions**
Tested on:
v3.15.1
V3.6.0

- Android: 9
- react-native-camera: v.3.15.1 and v3.6.0
- react-native: 5.9.10
- react: 16.8.3

**Description/Current Behaviour**

When recordAsync starts the camera starts freezing, and the video seems to be black and white

**Expected Behaviour**

**Steps to Reproduce**

Just using recordAsync causes this.

**Additionals**
https://i.imgur.com/RvOBbru.png

Some logs from Android Studio:

2020-01-27 12:15:50.441 2334-9972/? E/mm-camera: <MCT   ><ERROR> 790: mct_controller_proc_bus_msg_internal: MCT_BUS_MSG_SEND_HW_ERROR
2020-01-27 12:15:50.441 2334-2334/? E/mm-camera: <MCT   ><ERROR> 645: main: main: camera daemon notify error
2020-01-27 12:15:50.441 569-9974/? E/QCamera: <MCI><ERROR> mm_camera_event_notify: 189: Camera Event DAEMON DIED received
2020-01-27 12:15:50.443 7942-7942/com.streamer E/Camera: Error 100
2020-01-27 12:15:52.909 2334-10035/? E/mm-camera: <MCT   ><ERROR> 95: mct_bus_sof_thread_run: Session 1: Hinting SOF freeze to happen.         Sending event to dump info
2020-01-27 12:15:52.910 2334-10059/? E/mm-camera: <MCT   ><ERROR> 42: server_debug_dump_data_for_sof_freeze: Failure in opening /data/misc/camera/sof_freeze_dump.txt
2020-01-27 12:15:55.409 2334-10035/? E/mm-camera: <MCT   ><ERROR> 99: mct_bus_sof_thread_run: FATAL Session 1: SOF Freeze! Sending error message
2020-01-27 12:15:55.410 2334-9972/? E/mm-camera: <MCT   ><ERROR> 790: mct_controller_proc_bus_msg_internal: MCT_BUS_MSG_SEND_HW_ERROR
2020-01-27 12:15:55.411 2334-2334/? E/mm-camera: <MCT   ><ERROR> 645: main: main: camera daemon notify error
2020-01-27 12:15:55.411 569-9974/? E/QCamera: <MCI><ERROR> mm_camera_event_notify: 189: Camera Event DAEMON DIED received
2020-01-27 12:15:55.413 7942-7942/com.streamer E/Camera: Error 100

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2020-02-14:
Does the app crash? I doubt it is related to the camera library in any way, but rather google's own android camera library. You might want to try with the Camera2 api (`useCamera2` I believe), but that implementation is quite buggy and unstable to be honest.

leeopop, 2020-03-08:
@livnunes You may try ver 3.8.0.
If this solved your problem, I think you are facing same problem to me.

closed 2020-05-14 by stale[bot]