react-native-camera/react-native-camera #2693: iOS App Crash on Camera Page in Test-flight/Production build but works fine in development

priyasamyal, 2020-01-30:
# Bug Report

**To Do First**

- [ Yes] Did you try latest release?
- [Yes ] Did you try master?
- [ Yes] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android:
- iOS: 11.3
- react-native-camera: 3.16.0
- react-native: 0.61.4
- react: 16.9.0
  

**Description/Current Behaviour**

App crash in upload profile image of production build. The Code works perfection in development mode.

**Expected Behaviour**

Production build should work exactly the same way as it works in development build.
**Steps to Reproduce**

**Below is the prop for RNCamera**
   <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          captureAudio={false}
          style={styles.preview}
          type={this.state.cameraToggle}
          flashMode={this.state.flash}
          ratio="4:3"
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          androidRecordAudioPermissionOptions={{
            title: 'Permission to use audio recording',
            message: 'We need your permission to use your audio',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}>
I have attached the crash zip file from test flight.
[testflight_feedback (2).zip](https://github.com/react-native-community/react-native-camera/files/4133854/testflight_feedback.2.zip)

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

lucasluca, 2020-03-11:
Same issue here

closed 2020-05-17 by stale[bot]

Vikmanatus, 2021-05-09:
Hello to everyone,

I had the same issue today with the versions 3.4.3, 3.4.5, 3.4.6 on iPhone XSmax (IOS 14.4.2 & 14.5)
With the help of crashlitycs I saw that the stack trace of this error was the calling of:
 __41-[RNCamera initializeCaptureSessionInput]_block_invoke + 1588 (Not sure but I guess it is for Camera)
 
RNCamera.m - Ligne 1390
-[RNCamera initializeAudioCaptureSessionInput] + 1390(Not sure. but I guess it is for microphone)

The app was perfectly running on Android & IOS simulators but crashing in Testflight on real device

I had only the NSCameraUsageDescription and was only using the library for taking photo, but I solved it by adding the NSMicrophoneUsageDescription in the Info.plist and requesting microphone permission

I don't really know if this the attended behaviour of the library in Production builds or if this a bug, but I hope this answer will help someone =) 

Best Reguards