react-native-camera/react-native-camera #2695: Camera video recording stops when incoming call on iOS

nitinaloha, 2020-01-31:
# Bug Report
Camera video recording stops automatically when a incoming call is received on the iOS devices.

Can any one help us with this issue or any work around if any.

Below is the code we am using

`<RNCamera
                            ref={ref => {
                                this.camera = ref;
                            }}
                            style={styles.preview}
                            type={RNCamera.Constants.Type.back}
                            androidCameraPermissionOptions={null}
                            androidRecordAudioPermissionOptions={null}
                            defaultVideoQuality={RNCamera.Constants.VideoQuality["4:3"]}
                        />`

`this.setState({ recording: true, driveStart:currentDatFormat[0]});
                const options = { 
                    quality     : RNCamera.Constants.VideoQuality["4:3"],
                    videoBitrate: 0.8 * 1000 * 1000
                };
                // Get the device dimension
                const dim = Dimensions.get('window');

                if (Platform.OS === 'ios') {
                    const majorVersionIOS = parseInt(Platform.Version, 10);

                    if (majorVersionIOS >= 11) {
                        if (dim.width >= 800) {
                            options['codec'] = RNCamera.Constants.VideoCodec['HVEC'];
                        } else {
                            options['codec'] = RNCamera.Constants.VideoCodec['H264'];
                        }
                    } else {
                        options['codec'] = RNCamera.Constants.VideoCodec['H264'];
                    }
                }
                // Get camera cache data
                const promise = this.camera.recordAsync(options);
                const data = await promise;`

Thanks in advance

cristianoccazinsp, 2020-02-03:
Have you tried to record without audio? `captureAudio=false`? You cannot record video while on a phone call with audio, but I'm not sure if you can with audio. Keep in mind that anything that "pauses" the app will also pause video recording, so if you have to switch to another app to take the call, your app gets paused, and so does video.

nitinaloha, 2020-02-03:
Yes I have tried captureAudio=false but not worked.
What happens is in iPhone devices whenever an incoming call rings the phone app comes up and our app goes in background.
This behaviour stops the video recording since our app goes in background.
Please test this scenario and let me your feedback.

cristianoccazinsp, 2020-02-03:
If the app goes to background, you really can't do much about the video recording getting stopped. This is iOS doing it automatically and you cannot do anything to prevent it.  

What you would normally do, is to resume the video recording pointing to the same file once the app goes back to foreground; however, I'm not sure if RNCamera supports this. You can try setting the `path` prop to some known path and test if you can continue recording on the same file or not.

According to the docs: 

> path (file path on disk). Specifies the path on disk to record the video to. You can use the same uri returned to continue recording across start/stops

But I have never tested this.

nitinaloha, 2020-02-03:
I will try this option and update. Thank you for your insights.

nitinaloha, 2020-02-12:
> If the app goes to background, you really can't do much about the video recording getting stopped. This is iOS doing it automatically and you cannot do anything to prevent it.
>
> What you would [...]

I tried this solution but unfortunately this is not working.
The RNCamera npm also mentioned about path parameter that if "You can use the same uri returned to continue recording across start/stops", but still does not work.

Can someone help us here.

cristianoccazinsp, 2020-02-12:
What happens when you record on the same file/path? Does it create a whole new file that overwrites the previous one? Does it append? The docs seems to indicate that doing so would just append to the file, but perhaps that was never implemented.  Unless someone is able to implement that feature, I don't think there can be any other work around except taking multiple videos and merging them on a server.

nitinaloha, 2020-02-12:
"Starting video recording failed. Another recording might be in progress."
I get this error when I use the previous uri as the path parameter to resume video create a single file.

cristianoccazinsp, 2020-02-12:
Are you sure that error is for iOS? Seems to me that error string is for Android.

nitinaloha, 2020-02-12:
Yes you are correct that error is for Android.

Could you suggest any way to resume video recording or merge the two videos.

cristianoccazinsp, 2020-02-12:
If you are uploading to a server, you might want to look at your backend's platform a way to use `ffmpeg`. 

Merging this on the device will require native code, but I don't think it would be too hard. There's an outdated library here https://github.com/shahen94/react-native-video-processing  , but it could be a good start.

Other than that, I would look into implementing such feature within this library. It shouldn't be too complicated if you know some java/obj-c :)

nitinaloha, 2020-02-12:
Okay thank you for the suggestion I will go through and try some if they work.

closed 2020-04-19 by stale[bot]