react-native-camera/react-native-camera #2698: Camera stop Issue with new react-navigation behaviour - Android

cristianoccazinsp, 2020-02-07:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: Any
- iOS: None
- react-native-camera: 3.17.0
- react-native: 0.61.5
- react: 16.12.0
- react-navigation: 4.1.1
- react-navigation-stack: 2.1.1

**Description/Current Behaviour**
React navigation will now remove screens that are down in the stack (in my test, 2 pushed screens is enough to trigger this). When this happens, RNCamera receives a surface destroyed event, which in turn stops the camera. However, RNCamera is unable to "restore" the camera after the surface is re-created/updated because the camera is already fully stopped.

**Expected Behaviour**
RNCamera should be able to respond to surface destroy/update/create events and resume the camera if needed.

**Steps to Reproduce**
Mount a screen with a RNCamera component running with preview. Push 2 extra screens, then go back to the screen with the camera. Upon returning, the camera will no longer be work and a complete re-mount is needed.

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2020-02-07:
Related discussion here: https://github.com/react-navigation/stack/issues/407

sibelius, 2020-02-07:
I think camera should be robust enough to enable react-navigation mount and unmount screens

cristianoccazinsp, 2020-02-07:
I've just added a new commit to https://github.com/react-native-community/react-native-camera/pull/2685 that I believe will resolve the issue. This took me for surprise since we never had this issue before with previous versions of react-navigation. But you are right, the camera should be able to detect if its surface has been destroyed or not and react accordingly. I think the above PR will solve it.

sibelius, 2020-04-07:
is this solved?

cristianoccazinsp, 2020-04-07:
Yes. Actually, react-navigation reverted their Android change so it is no longer an issue with that library. However, the android fix was still applied in case another library/component uses the same behaviour (removeClippedSubviews).

closed 2020-06-13 by stale[bot]

ghasemikasra39, 2021-06-23:
getting the same issue on ios