react-native-camera/react-native-camera #2700: FPS changes without changing the code.

cinjon, 2020-02-09:
# Bug Report

I am using iOS and have set the recording options as:

`
{ 
quality: RNCamera.Constants.VideoQuality['1080p'], 
codec: RNCamera.Constants.VideoCodec.H264
}
`

The result is that *sometimes* the videos have FPS of 15.01 as reported by running `ffmpeg -i /path/to/video`. But this doesn't always happen. Is there some place where settings are being applied based on certain conditions? I'm not very familiar with native development and so I am having trouble figuring out where in the native code this issue could be happening.

**Platforms**

**Versions**

- Android:
- iOS: iPad
- react-native-camera: 3.15.1
- react-native: 0.61.5
- react: 16.9.0
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2020-02-10:
I don't think the camera code has anything to do. This is most likely something from iOS or the camera itself, or the device doing too much work and not being able to keep up with the processing. 

Have you tried with a lower resolution, or another codec? Does the same happen with the native camera?

cinjon, 2020-02-10:
Thanks for the response. I have not tried another resolution or codec in the wild. And it does not happen with the native camera. How would I be able to tell if the processing is too much?

cristianoccazinsp, 2020-02-10:
You could try with an attached debugger and XCode. You will be able to see very nice RAM and CPU numbers when doing that.

cinjon, 2020-02-10:
Is there a way to do that in the wild (i.e. w/o attached debugger) and dump to file somewhere?

cristianoccazinsp, 2020-02-10:
Not that I'm aware of :)

cinjon, 2020-02-12:
Unclear what was happening here, but if I change it to HEVC, then it doesn't happen.

cristianoccazinsp, 2020-02-12:
What device were you testing with? That definitely sounds like the phone is unable to keep up recording at such high quality. HEVC is more efficient (and apple's default), so it would make sense.

cinjon, 2020-02-12:
I have been testing with an iPad 6th Gen (10.2 in) 13.3.1.

closed 2020-04-19 by stale[bot]