react-native-camera/react-native-camera #2701: Error: Cannot Record

izzykreie, 2020-02-13:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

I've installed all of the libraries and dependencies my camera needs to run on react native, but I get this warning: 

Error Domain=AVFoundationErrorDomain Code=-11814 "Cannot Record"

Any ideas on how to fix? I followed the documentation to the T and looked through all the Q&A's.

**Tags**
<!react-native-->

cristianoccazinsp, 2020-02-13:
Are you testing on a real device? Can you debug with XCode and get the actual error?

izzykreie, 2020-02-13:
@cristianoccazinsp I'm testing on an emulator. I've just been using visual studio up to this point to edit the code but yea I can use Xcode, I'm going to figure out how to import my project

cristianoccazinsp, 2020-02-13:
As far as I know, recording does not work on a simulator. You will need a real device.

closed 2020-04-23 by stale[bot]