react-native-camera/react-native-camera #2703: IOS - Stop when FaceDetection starts recording.

aksmfosef11, 2020-02-14:
# Bug Report

**To Do First**

- [ x ] Did you try latest release?
- [ x ] Did you try master?
- [ x ] Did you look for existing matching issues?

**Platforms**

**Versions**

- iOS:12.3.1 , 13.2.2
- react-native-camera: 3.18.0
- react-native: 0.61.5
- react:16.9.0

**Description/Current Behaviour**

Stop when FaceDetection starts recording.

**Expected Behaviour**

Stop when FaceDetection starts recording.

**Steps to Reproduce**

I downloaded and ran mlKit from examples.
If you start recording with FaceDetection running, the FaceDetection immediately before recording starts is frozen.

ysrarff, 2020-03-30:
Hi @aksmfosef11,

I needed this to be my requirements a few months back and it seems that it is not supported yet. The face detection (or any of the ML features) can only carry out its process when it is in preview mode.

As a workaround, I had to dismiss the realtime detection feature and resort to making it a post-recording process, by converting the video into individual frames (10 frames per second) via ffmpeg and running the firebase face detection module as a service on the cloud.

If u still insist on running it locally on the user's device then u could try and use [react-native-opencv](https://github.com/adamgf/react-native-opencv3). Although it requires a lot of native code setup/configurations. I managed to get it working but the camera was stuttering as it was importing too many ML models, and at that point I dropped the whole thing altogether as it was time-consuming for me (I don't have any Swift/Obj-C experience),

Really wish, they would work on this as well... Cheers!

lucastonon, 2020-05-23:
> Hi @aksmfosef11,
> 
> I needed this to be my requirement a few months back and it seems that it is not supported yet. The face detection (or any of the ML features) can only carry out its process when it is in preview mode.
> 
> As a workaround, I had to dismiss the realtime detection feature and resort to making it a post-recording process, by converting the video into individual frames (10 frames per second) via ffmpeg and running the firebase face detection module as a service on the cloud.
> 
> If u still insist on running it locally on the user's device then u could try and use [react-native-opencv](https://github.com/adamgf/react-native-opencv3). Although it requires a lot of native code setup/configurations. I managed to get it working but the camera was stuttering as it was importing too many ML models, and at that point I dropped the whole thing altogether as it was time-consuming for me (I don't have any Swift/Obj-C experience).
> 
> Really wish, they would work on supporting this kind of requirement in the future. Cheers!

I have a similar requirement, and I might try opencv. Do you care to tell me in which device was it stuttering?

lucastonon, 2020-05-23:
@fabriziobertoglio1987 @SimonErm @MateusAndrade @itsjgf @friederbluemle @TomSwift 
Sorry to tag you all, but since this issue has been going on for some months, can you guys just point us to the right direction? Do you know if this is a technical limitation or is it doable on react-native-camera?

Thank you so much!

closed 2020-07-30 by stale[bot]