react-native-camera/react-native-camera #2705: java.lang.NoSuchMethodError : No virtual method setExposureCompensation(F)V

kernoeb, 2020-02-16:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Description/Current Behaviour**

Hello,
When I'm trying to use react-native-camera, it crashes everytime.
I tried a lot of code examples, etc...
This is the error (from adb logcat) 

`02-16 22:12:32.103 32077 32077 E unknown:ViewManager: Error while updating prop exposure
02-16 22:12:32.103 32077 32077 E unknown:ViewManager: java.lang.reflect.InvocationTargetException
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at java.lang.reflect.Method.invoke(Native Method)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:83)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:134)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:54)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:48)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:278)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:179)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:773)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:888)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue.access$2200(UIViewOperationQueue.java:42)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:948)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:28)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:174)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.view.Choreographer$CallbackRecord.run(Choreographer.java:947)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.view.Choreographer.doCallbacks(Choreographer.java:761)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.view.Choreographer.doFrame(Choreographer.java:693)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:935)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.os.Handler.handleCallback(Handler.java:873)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.os.Handler.dispatchMessage(Handler.java:99)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.os.Looper.loop(Looper.java:193)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at android.app.ActivityThread.main(ActivityThread.java:6923)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at java.lang.reflect.Method.invoke(Native Method)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:870)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager: Caused by: java.lang.NoSuchMethodError: No virtual method setExposureCompensation(F)V in class Lorg/reactnative/camera/RNCameraView; or its super classes (declaration of 'org.reactnative.camera.RNCameraView' appears in /data/app/com.memento-gfMWw9rBHG1oMH9oiKe_Nw==/base.apk!classes2.dex)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   at org.reactnative.camera.CameraViewManager.setExposureCompensation(CameraViewManager.java:95)
02-16 22:12:32.103 32077 32077 E unknown:ViewManager:   ... 25 more
02-16 22:12:32.104 32077 32077 E unknown:ReactNative: Exception in native call
02-16 22:12:32.104 32077 32077 E unknown:ReactNative: com.facebook.react.bridge.JSApplicationIllegalArgumentException: Error while updating property 'exposure' of a view managed by: RNCamera
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:98)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:134)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:54)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:48)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:278)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:179)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:773)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:888)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.UIViewOperationQueue.access$2200(UIViewOperationQueue.java:42)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:948)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:28)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:174)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.view.Choreographer$CallbackRecord.run(Choreographer.java:947)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.view.Choreographer.doCallbacks(Choreographer.java:761)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.view.Choreographer.doFrame(Choreographer.java:693)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:935)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.os.Handler.handleCallback(Handler.java:873)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.os.Handler.dispatchMessage(Handler.java:99)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.os.Looper.loop(Looper.java:193)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at android.app.ActivityThread.main(ActivityThread.java:6923)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at java.lang.reflect.Method.invoke(Native Method)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:870)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative: Caused by: java.lang.reflect.InvocationTargetException
[... cut ...] View' appears in /data/app/com.memento-gfMWw9rBHG1oMH9oiKe_Nw==/base.apk!classes2.dex)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   at org.reactnative.camera.CameraViewManager.setExposureCompensation(CameraViewManager.java:95)
02-16 22:12:32.104 32077 32077 E unknown:ReactNative:   ... 25 more
02-16 22:12:32.114 32077 32077 D AndroidRuntime: Shutting down VM
`

There's some errors about "exposure", don't know how to fix this :(

Thanks !

**Steps to Reproduce**

Just use the library..

gavinjkw, 2020-02-21:
I am also experiencing this issue.

AadityaDev, 2020-02-22:
I am also facing the same error.

cristianoccazinsp, 2020-02-24:
Are you setting the exposure value somewhere? Looks like a build/dex issue.

gavinjkw, 2020-02-24:
I am not setting exposure. It seems like its breaking on view.setExposureCompensation(exposure).  

@ReactProp(name = "faceDetectorEnabled")
  public void setFaceDetecting(RNCameraView view, boolean faceDetectorEnabled) {
    view.setShouldDetectFaces(faceDetectorEnabled);
  }

cristianoccazinsp, 2020-02-24:
Maybe share some code, lib info, RN info, device and OS info, etc.

vladimirsvsv77, 2020-02-25:
I have the same error.

vladimirsvsv77, 2020-02-25:
I solved it for my case. Before using RNCamera, I used expo-camera, then I've removed expo-camera, and now RNCamera works.  

`yarn remove expo-camera`

closed 2020-05-02 by stale[bot]

samaneh-kamalian, 2020-11-29:
I have same issue in android 4.1.2
:(