react-native-camera/react-native-camera #2706: OnePlus A5000, Android 9, API 28

Baskerville42, 2020-02-17:
# Bug Report

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9, API 28
- react-native-camera: 3.17.0
- react-native: 0.60.6
- react: 16.8.6
- react-navigation: 1.0.0-beta.21

**Description/Current Behaviour**

This bug related only to one device with specific Android version. When I trying to record a video by using `recordAsync`, after the first seconds of recording I see a white screen. In Logcat I can see next logs:

```
2020-02-17 11:36:52.365 13248-13359/app.nowat.android W/unknown:ViewManagerPropertyUpdater: Could not find generated setter for class com.lwansbrough.RCTCamera.RCTCameraViewManager
2020-02-17 11:36:52.367 13248-13359/app.nowat.android W/unknown:ViewManagerPropertyUpdater: Could not find generated setter for class org.reactnative.camera.CameraViewManager
2020-02-17 11:37:19.352 13248-13752/app.nowat.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:19.352 13248-13752/app.nowat.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:19.612 13248-13752/app.nowat.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:19.612 13248-13752/app.nowat.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:19.683 13248-13752/app.nowat.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:19.683 13248-13752/app.nowat.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:22.737 13248-13752/app.nowat.android E/CAMERA_1::: Record setParameters failed
    java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:2442)
        at com.google.android.cameraview.Camera1.record(Camera1.java:752)
        at com.google.android.cameraview.CameraView.record(CameraView.java:610)
        at org.reactnative.camera.RNCameraView.access$2001(RNCameraView.java:31)
        at org.reactnative.camera.RNCameraView$3.run(RNCameraView.java:285)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2020-02-17 11:37:22.737 13248-13752/app.nowat.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:22.737 13248-13752/app.nowat.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:22.749 13248-13752/app.nowat.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:22.749 13248-13752/app.nowat.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:23.955 13248-13752/app.nowat.android E/Camera: Error 2
2020-02-17 11:37:24.187 13248-13531/app.nowat.android W/CameraBase: Camera service died!
2020-02-17 11:37:24.193 13248-13497/app.nowat.android W/CameraBase: mediaserver's remote binder Camera object died
2020-02-17 11:37:24.230 13248-13752/app.nowat.android E/CAMERA_1::: stopMediaRecorder stop failed
    java.lang.RuntimeException: stop failed.
        at android.media.MediaRecorder.stop(Native Method)
        at com.google.android.cameraview.Camera1.stopMediaRecorder(Camera1.java:1420)
        at com.google.android.cameraview.Camera1.stopRecording(Camera1.java:770)
        at com.google.android.cameraview.Camera1.onError(Camera1.java:1471)
        at android.media.MediaRecorder$EventHandler.handleMessage(MediaRecorder.java:1450)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2020-02-17 11:37:24.238 13248-13752/app.nowat.android E/Camera: Error 100
2020-02-17 11:37:26.023 13248-13248/app.nowat.android I/MediaPlayer: setDataSource:file:///data/user/0/app.nowat.android/cache/Camera/a9e76a7c-5d53-4236-b14f-6421c647aca1.mp4
2020-02-17 11:37:26.089 13248-13248/app.nowat.android I/MediaPlayer: setDataSource:file:///data/user/0/app.nowat.android/cache/Camera/a9e76a7c-5d53-4236-b14f-6421c647aca1.mp4
```

**Expected Behaviour**

The video should be recorded on any device

**Steps to Reproduce**

There is my code which not working: [https://gist.github.com/Baskerville42/608d00615e8707281051c1b1534ede53](https://gist.github.com/Baskerville42/608d00615e8707281051c1b1534ede53)

cristianoccazinsp, 2020-02-24:
Does the device actually crash? I've looked that error log in the source code, and it is only a log, but the exception is handled.

The other errors below after recording starts seems strange:
```
2020-02-17 11:37:22.737 13248-13752/app.myapp.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:22.737 13248-13752/app.myapp.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:22.749 13248-13752/app.myapp.android D/Camera: CAMERA_MSG_STATE_CALLBACK
2020-02-17 11:37:22.749 13248-13752/app.myapp.android D/Camera: mCameraStateCallback is null
2020-02-17 11:37:23.955 13248-13752/app.myapp.android E/Camera: Error 2
2020-02-17 11:37:24.187 13248-13531/app.myapp.android W/CameraBase: Camera service died!
2020-02-17 11:37:24.193 13248-13497/app.myapp.android W/CameraBase: mediaserver's remote binder Camera object died
```

Looks like the camera is misbehaving.

You could try searching in the source code (Camera1.java) for `Record setParameters failed"` and commenting the lines around it to see if that's causing issues.  

Lastly, you could try with version 3.16.x or 3.15.x just in case the bug was introduced afterwards with some of the updates to prevent ANRs.

Device specific issues are quite hard to debug without the actual device.

Baskerville42, 2020-02-25:
@cristianoccazinsp thanks for your reply. I forgot to update this issue. The problem was with the unsupported resolution for this device - **480p**. After I changed the resolution to 720p it's all right.
How do you think, is it possible to make the method which will return all available resolutions for a specific device?

cristianoccazinsp, 2020-02-25:
I think there's one already, but I don't have it at hand right now; check
the docs.

fabriziobertoglio1987, 2020-03-02:
I [quote][1]

>If nothing is passed the device's highest camera quality will be used as default. Note: This solve the flicker video recording issue for iOS

[1]: https://react-native-community.github.io/react-native-camera/docs/rncamera#defaultvideoquality

closed 2020-05-09 by stale[bot]