react-native-camera/react-native-camera #2707: Using mlkit in ios

IrmantasVaracinskas, 2020-02-17:
# Question

**Ask your Question**
I am trying to add GoogleVision api for barcode scanning. Created firebase project, added `GoogleService-Info.plist`, added `pod 'Firebase/Core'`. But when trying to add  
`pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'BarcodeDetectorMLKit'
  ]`  
and running `pod install` it shows error `target has transitive dependencies that include statically linked binaries: (FirebaseCore, FirebaseCoreDiagnostics, GoogleDataTransportCCTSupport, GoogleDataTransport, and FirebaseInstanceID)`.  
Is there something that I'm doing wrong or something is not updated?

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2020-02-17:
can you run examples/mlkit example?

IrmantasVaracinskas, 2020-02-17:
Yes, it seems that example is running and working correctly... Maybe there are some version mismatch in my application...

closed 2020-02-17 by sibelius

IrmantasVaracinskas, 2020-02-17:
P.s. This issue only occurs when using `use_frameworks!` in podfile

rexlow, 2020-02-26:
I am too having this issue and removing `use_frameworks!` does not solve this issue.

sbycrosz, 2020-05-28:
Hello!

We've also run into this problem.

What do you think about adding `s.static_framework = true` in react-native-camera's Podspec file since react-native-camera depends on a static libraries (Firebase)? [(Theoretically shouldn't break anything)](https://stackoverflow.com/questions/48728317/can-enabling-static-framework-in-podspec-break-existing-apps)

This will allow users to keep `use_framework!` on their Podfile :)

Cheers,
Sam