react-native-camera/react-native-camera #2709: Front Camera Preview Zoom Issue

hejsfj, 2020-02-20:
# Bug Report

**Platforms**

iOS 

**Versions**

- iOS: 13.3.1
- react-native-camera: 3.18.0
- react-native: 0.61.5

**Description/Current Behaviour**

When using the front camera the preview image of the camera is zoomed in. Once I hit record the image zooms out a bit and starts recording. Once I end the recording the image is again zoomed in a bit. 

Has anyone experience this behaviour before?

**Expected Behaviour**

Same "zoom" when in preview mode and record mode

**Steps to Reproduce**

`import React, { useRef, useCallback, useState } from 'react';
import { SafeAreaView } from 'react-navigation';
import { View, TouchableOpacity, Text, StyleSheet } from 'react-native';
import { RNCamera } from 'react-native-camera';
import { useWithStyles } from '../../../theming/useWithStyles';
import { Header, HeaderBackButton, HYButton } from '../../../components';
import { useNavigation } from 'react-navigation-hooks';

const styles = ({ theme }) => ({
  camera: {
    // flex: 1,
    ...StyleSheet.absoluteFillObject,
  },
});

const VIDEO_QUALITY = RNCamera.Constants.VideoQuality['720p'];
const VIDEO_STABILIZATION_MODE = RNCamera.Constants.VideoStabilization.auto;

export const VideoScreen = props => {
  const { classes } = useWithStyles(styles);
  const [isRecording, setIsRecording] = useState(false);
  const [cameraType, setCameraType] = useState(RNCamera.Constants.Type.front);
  const navigation = useNavigation();
  const cameraRef = useRef(null);

  const startRecording = useCallback(async () => {
    if (cameraRef) {
      const options = {
        quality: VIDEO_QUALITY,
      };
      setIsRecording(true);
      const video = await cameraRef.current.recordAsync(options);
      setIsRecording(false);
      if (video.isRecordingInterrupted) {
        //do not save the video!
      }
      console.log(video.uri);
    }
  }, [cameraRef]);

  const stopRecording = useCallback(() => {
    if (cameraRef) {
      cameraRef.current.stopRecording();
    }
  });

  const toggleCameraType = useCallback(() => {
    setCameraType(
      cameraType === RNCamera.Constants.Type.front
        ? RNCamera.Constants.Type.back
        : RNCamera.Constants.Type.front,
    );
  }, [cameraType]);

  return (
    <View style={{ flex: 1 }}>
      <RNCamera
        ref={cameraRef}
        style={classes.camera}
        type={cameraType}
        defaultVideoQuality={VIDEO_QUALITY}
        videoStabilizationMode={VIDEO_STABILIZATION_MODE}
        // androidCameraPermissionOptions={{
        //   title: 'Permission to use camera',
        //   message: 'We need your permission to use your camera',
        //   buttonPositive: 'Ok',
        //   buttonNegative: 'Cancel',
        // }}
        // androidRecordAudioPermissionOptions={{
        //   title: 'Permission to use audio recording',
        //   message: 'We need your permission to use your audio',
        //   buttonPositive: 'Ok',
        //   buttonNegative: 'Cancel',
        // }}
        // onGoogleVisionBarcodesDetected={({ barcodes }) => {
        //   console.log(barcodes);
        // }}
      />
      <SafeAreaView
        style={{ flex: 1, ...StyleSheet.absoluteFillObject }}
        forceInset={{ vertical: 'always', horizontal: 'always' }}>
        <Header
          left={
            <HeaderBackButton
              onPress={() => navigation.goBack()}
              customIconProps={{ color: 'white' }}
            />
          }
        />
        <View
          style={{ flex: 1, justifyContent: 'flex-end', alignItems: 'center' }}>
          <View style={{ flexDirection: 'row', alignSelf: 'stretch' }}>
            <View style={{ flex: 1 }} />
            <View
              style={{
                flex: 1,
                alignItems: 'center',
                justifyContent: 'center',
              }}>
              <TouchableOpacity
                style={{
                  backgroundColor: isRecording ? 'red' : 'white',
                  width: 80,
                  height: 80,
                  borderRadius: 40,
                }}
                onPress={!isRecording ? startRecording : stopRecording}
              />
            </View>
            <View
              style={{
                flex: 1,
                alignItems: 'center',
                justifyContent: 'center',
              }}>
              <TouchableOpacity
                style={{
                  backgroundColor: isRecording ? 'red' : 'white',
                  width: 60,
                  height: 60,
                  borderRadius: 60,
                }}
                onPress={toggleCameraType}
              />
            </View>
          </View>
        </View>
      </SafeAreaView>
    </View>
  );
};
`

cristianoccazinsp, 2020-02-24:
If recording and preview use different qualities/presets (which I believe is the default), then the preview and recording will look different, at least on iOS. You may see the same thing when using the native camera app. This is because photos and video use different resolutions. You could try playing around with `pictureSize` (iOS only) and `quality` when recording.

TStrauch, 2020-03-24:
Interestingly setting the pictureSize resolved it for the front camera on my end.
However, for me the issue still exists when using the back camera.

qja0707, 2020-04-14:
I have the same problem. The preview is zoomed out when I take a video

eraykose, 2020-04-19:
My recording `quality` options were '720p'. I set the `pictureSize` to "1280x720" and it solved the problem on IOS.

qja0707, 2020-04-21:
@eraykose Thanks a lot. I tried your solution, and it works. I use two quality options '1080p, 720p'. When I set `pictureSize` to "1280x720" with 1080p, camera is flickered but "1920x1080" is works well with both options.

closed 2020-06-27 by stale[bot]