react-native-camera/react-native-camera #2710: Video not recording when camera flip to back.

Nithanpandi, 2020-02-21:
First I tried to record video using back camera then I flip my camera and record video using front camera, it works fine. But after took video using front camera, I can't record videos using back camera. 
Also same issue happens when I try to record video using back camera, more then one time.

***This problem occurs in build apk, works in development.***

I'm using react-hooks and here is my sample code,
```
const cameraRef = useRef(null);
const [frontCam, setFrontCam] = useState(false);
const [isRecording, setRecording] = useState(false);

const takeVideo = async () => {
        const promise = cameraRef.current.recordAsync();
        setRecording(true);
        const data = await promise;
}

const stopVideo = async () => {
    await cameraRef.current.stopRecording();
  }

return(
<View>
 <RNCamera
          ref={cameraRef}
          style={styles.camera}
          type={
            frontCam ?
              RNCamera.Constants.Type.front
              : RNCamera.Constants.Type.back
          }
        /> 
      <TouchableOpacity
          onPress={() => setFrontCam(!frontCam)}
          style={styles.flip}>
          <Icon name="camera-party-mode" size={30} color="#ffffff" />
        </TouchableOpacity>

       {isRecording ?
          <Icon
            style={styles.record}
            onTouchStart={() => stopVideo()}
          />
          :
          <Icon
            style={styles.record}
            onTouchStart={() => takeVideo()}
          />
        }

</View>
);
```
"react-native": "0.61.5",
"react-native-camera": "^3.16.0"

cristianoccazinsp, 2020-02-24:
Are you recording right after switching cameras? Perhaps the camera is not ready? I haven't seen this happen, but you might want to listen to camera ready events before starting a new recording.

Nithanpandi, 2020-02-25:
@cristianoccazinsp  

Yes I tried recording after camera is ready. But facing same issue. This works fine during development, issue happens in build apk.

closed 2020-05-02 by stale[bot]