react-native-camera/react-native-camera #2716: Why does cameraPreview need to be active before taking a picture?

LouiseBC, 2020-02-26:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I have `pauseAfterCapture` enabled, but it's quite slow. So if you take a picture and then move the camera you can end up taking a picture with motion blur, or of a different subject. On iOS I can fix this by calling `.pausePreview()` myself before taking a picture, but this breaks on Android. I can see it is related to [this line](https://github.com/react-native-community/react-native-camera/blob/c7e92b29c8106acedfdc8d3d9b1ac7702895fdc6/android/src/main/java/com/google/android/cameraview/Camera1.java#L676), but it doesn't explain why this is necessary. Is this something that could be removed and/or fixed?

cristianoccazinsp, 2020-02-27:
For Android, it seems like it is a requirement:

`Important: Call startPreview() to start updating the preview surface. Preview must be started before you can take a picture.`
`After taking a picture, preview display will have stopped. To take more photos, call startPreview() again first.`

https://developer.android.com/reference/android/hardware/Camera

LouiseBC, 2020-02-27:
Good find, thanks. Perhaps we can look into why it takes so long until .pausePreview() is finally called. Judging by the amount of similar issues that were opened I know I'm not the only one struggling with this

closed 2020-05-04 by stale[bot]